adafruit/circuitpython #7904: displayio.bitmap "Space" characther given ValueError

jposada202020, 2023-04-24:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.0-62-g4986ad6d6-dirty on 2023-03-30; Adafruit PyPortal Titano with samd51j20

and

Adafruit CircuitPython 8.1.0-beta.1-36-g7d02bff6b on 2023-04-20; Adafruit Feather RP2040 DVI with rp2040
```

### Code/REPL

```python
import board
import displayio
from adafruit_bitmap_font import bitmap_font
from adafruit_display_text import label, bitmap_label

display = board.DISPLAY
main_group = displayio.Group()
MEDIUM_FONT = bitmap_font.load_font("fonts/LeagueSpartan-Bold-16.bdf")

bitmap = displayio.Bitmap(display.height, display.width, 1)
palette = displayio.Palette(2)
palette[0] = 0xFFFFFF

some_text = label.Label(
    MEDIUM_FONT,
    text="CircuitPython",
    x=display.width // 2,
    y=display.height // 2,
)
main_group.append(some_text)
display.show(main_group)

print("This is good")

some_text2 = label.Label(
    MEDIUM_FONT,
    text="Circuit Python",
    x=display.width // 3,
    y=display.height // 3,
)
main_group.append(some_text2)
display.show(main_group)
```

### Behavior

```python
>>> import t
This is good
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "t.py", line 29, in <module>
  File "/lib/adafruit_display_text/label.py", line 100, in __init__
  File "/lib/adafruit_display_text/label.py", line 424, in _reset_text
  File "/lib/adafruit_display_text/label.py", line 281, in _update_text
  File "/lib/adafruit_bitmap_font/glyph_cache.py", line 54, in get_glyph
  File "/lib/adafruit_bitmap_font/bdf.py", line 191, in load_glyphs
ValueError: width must be 1-32767

```

### Description

As showed in the bug code the text gives a ValueError. This used to work as in the advance example (aka test suite) for the library... here https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/blob/main/examples/display_text_advance_example.py

I used the example for this PR
https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/pull/181

### Additional information

Not sure, maybe we change for checking the value of the bitmap. the number sounds familiar but I did not look in the core for clues

RetiredWizard, 2023-04-25:
I think this may have been fixed by #7867 which was merged on 4/20. I tested using the Feather RP2040 DVI using a build from 6/21 and didn't see the issue. Can you try the latest build and retest?

jposada202020, 2023-04-25:
Thanks.  @RetiredWizard ill try tomorrow. Day pystack ran out 😆

jposada202020, 2023-04-25:
That made it. Thank you @RetiredWizard  :)

closed 2023-04-25 by jposada202020