adafruit/circuitpython #7906: Pico W sometimes won't start AP immediately after stopping Station

anecdata, 2023-04-25:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.1-46-gb5a26c7f9 on 2023-04-24; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import os
import traceback
import microcontroller
import supervisor

time.sleep(3)

while True:
    try:
        print(f"starting station...", end=" ")
        wifi.radio.start_station()
        while not wifi.radio.connected:
            wifi.radio.connect(os.getenv("WIFI_SSID"), os.getenv("WIFI_PASSWORD"))
            print(f"connected={wifi.radio.connected} {wifi.radio.ipv4_address}")
        print(f"stopping station...")
        wifi.radio.stop_station()  # no delay between stop station stop and start AP
        wifi.radio.start_ap('test', 'testtest')
        print(f"ap_active={wifi.radio.ap_active} {wifi.radio.ipv4_address_ap}")
        time.sleep(15)
        microcontroller.reset()
    except Exception as ex:
        print(f"")
        traceback.print_exception(ex, ex, ex.__traceback__)
```

### Behavior

This arose on Discord (). About 10% of the time, `wifi.radio.start_ap()` will fail with `RuntimeError: Wifi is in station mode.` if it immediately follows `wifi.radio.stop_station()`.

A small time delay, or:
```py
        wifi.radio.stop_station()
        while wifi.radio.connected:
            pass
        wifi.radio.start_ap('test', 'lksdfjhksdfjlsdfjgh')
```
seems to suffice to let start_AP() succeed consistently.

Is this the solution, or could `wifi.radio.stop_station()` not return until the station is fully stopped?

dhalbert, 2023-05-15:
That makes sense. If the delay is known, the code can wait. It could also poll some status itself and return quicker if it has stopped, but that may not be possible. It shouldn't poll indefinitely in case the status is never satisfied.

anecdata, 2023-05-15:
How about looping this check:
https://github.com/adafruit/circuitpython/blob/827eaeb1f9e0d3711215cfeb6558ac64deda7fea/ports/raspberrypi/common-hal/wifi/Radio.c#L173
in `wifi.radio.stop_station()`, so that when it exits, it's a clean stop?

bill88t, 2023-05-15:
Sounds good on paper, but will have to be tested. I will do so.
I will also add a regular timeout so it doesn't get stuck.