adafruit/circuitpython #7910: Union type for 'any kind of display-like object'

jepler, 2023-04-25:
During PyCon sprints, a participant raised the issue of needing to type an argument as 'any kind of displayio display'. It would be nice if the core (or maybe circuitpython-typing?) had a Union type that included all these types, for libraries to refer to.

tekktrik, 2023-04-28:
`circuitpython_typing` could add a `typing.Protocol` for displays (or multiple kinds) if we want to duck type it.  Can you link to an example of where it would be used to know what methods/attributes are required?

jepler, 2023-04-28:
I didn't take note of which repo the author was working on. However, `adafruit_bitmapsaver` looks like one such:

```
def save_pixels(                                                                
    file_or_filename: Union[str, BufferedWriter],                               
    pixel_source: Union[Display, Bitmap] = None,                                
    palette: Optional[Palette] = None,                                          
) -> None:                                                                      
```