adafruit/circuitpython #7911: DVI resolution is actually 320x240, not 640x480

todbot, 2023-04-26:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.1-48-gabb0d5e62 on 2023-04-24; Adafruit Feather RP2040 DVI with rp2040
```

### Code/REPL

```python
import board
display = board.DISPLAY
print("display width, height:", display.width, display.height)
```

### Behavior

Prints out:
```
code.py output:
display width, height: 320 240
```

### Description

I am not sure that 640x480 is actually possible. Most all the PicoDVI examples I've seen have been 320x240. 

### Additional information

Trying to create a duplicate of `board.DISPLAY` using `picodvi.Framebuffer` yields "invalid height" when trying to do 320x240x8 or "MemoryError: memory allocation failed" when trying to do 640x480x2 or 640x480x4.

dglaude, 2023-04-26:
Hi,

I suggest you look at the code in this issue for a way to get a 640x480x2 display:
https://github.com/adafruit/circuitpython/issues/7898
(just ignore the afb and the circle, but the rest of the code should work)

I confirm most resolution and number of colours do fail on the Memory side.
But I manadged to go in "High Res" mode and that was my only goal.
And is why I wanted to write to the framebuffer to avoid alocating any extra bit of memory.

I also confirm that having width height match the framebuffer resolution (and not the visible resolution) is super important.
This issue show how confusing it can be:
https://github.com/adafruit/circuitpython/issues/7894

I hope it help, please check what further help you need, there are things to do, but we might want to be very specific.
It will be much easier with a learn guide... or some tips from todbot. :-)

Regards

dglaude, 2023-04-26:
I have put together a demo or high-resolution DVI output.

It create a bitmap as big as the frame buffer, so it is very inefficient.

This works for 640x480 black and white.
It seems ok in 800x480 too.
But it fail if you want to go grey scale with 2 bits per pixel.

Also, most "turtle" attempt will fail as turtle reserve a frame buffer (I guess).

So either we get an addressable frame buffer from picodvi and can reuse that.
Or you have to draw with object in displayio, but don't expect to be able to address any pixel you want, the way you want.

At least most of my attempt have failed on memory usage.

Here is the working code:
```
#High Res 640x480x1

import displayio
displayio.release_displays()

import time
import board
import picodvi
import framebufferio
from bitmaptools import draw_line

w=640
h=480
cd=1

fb = picodvi.Framebuffer(
    width=w, height=h, color_depth=cd,
    clk_dp=board.CKP, clk_dn=board.CKN,
    red_dp=board.D0P, red_dn=board.D0N,
    green_dp=board.D1P, green_dn=board.D1N,
    blue_dp=board.D2P, blue_dn=board.D2N)

display = framebufferio.FramebufferDisplay(fb)

print(board.DISPLAY.width, board.DISPLAY.height)
print(display.width, display.height)

group = displayio.Group()

palette = displayio.Palette(1)
palette[0] = 0xFFFFFF

a2 = displayio.Group(x=0, y=0, scale=1)
a2bitmap = displayio.Bitmap(display.width, display.height, 2)

tile2 = displayio.TileGrid(a2bitmap, pixel_shader=palette, x=0, y=0)

draw_line(a2bitmap, 1, 1, 1, display.height-2, 1)
draw_line(a2bitmap, 1, 1, display.width-2, 1, 1)
draw_line(a2bitmap, display.width-2, 1, display.width-2, display.height-2, 1)
draw_line(a2bitmap, 1, display.height-2, display.width-2, display.height-2, 1)

draw_line(a2bitmap, 1, 1, display.height-2, display.height-2, 1)

a2.append(tile2)
group.append(a2)

display.show(group)

while True:
    time.sleep(1)
```

todbot, 2023-04-26:
Sorry if I wasn't clear. What I mean to say is that the framebuffer setup code and docs are mistaken.  The [Feather RP2040 DVI `board.c` code]  thinks it's making a 640x480x8 display but is actually making a 320x240x8 display. See: [board.c#36](https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/boards/adafruit_feather_rp2040_dvi/board.c#L36):
```c
void board_init(void) {
    picodvi_framebuffer_obj_t *fb = &allocate_display_bus()->picodvi;
    fb->base.type = &picodvi_framebuffer_type;
    common_hal_picodvi_framebuffer_construct(fb, 640, 480,
        &pin_GPIO17, &pin_GPIO16,
        &pin_GPIO19, &pin_GPIO18,
        &pin_GPIO21, &pin_GPIO20,
        &pin_GPIO23, &pin_GPIO22,
        8);
```

And [the readthedocs documentation](https://docs.circuitpython.org/en/latest/shared-bindings/picodvi/index.html#picodvi.Framebuffer), indicate that the only allowable resolutions are 640x480 and 800x480 and implying that 640x480x16 is an achievable resolution. I am pretty sure this is incorrect.

tannewt, 2023-04-26:
> Trying to create a duplicate of `board.DISPLAY` using `picodvi.Framebuffer` yields "invalid height" when trying to do 320x240x8 or "MemoryError: memory allocation failed" when trying to do 640x480x2 or 640x480x4.

Please post this code. x4 isn't supported.

These are the supported arguments and the resulting framebuffer. The larger versions will be hard to allocate from within CP but may be possible from `board_init()`.

Grayscale:
* 640x480x1 -> 640x480x1
* 800x480x1 -> 800x480x1
* 640x480x2 -> 640x480x2
* 800x480x2 -> 800x480x2

Monochrome:
* 640x480x8 -> 320x240x8 (default on feather)
* 800x480x8 -> 400x240x8
* 640x480x16 -> 320x240x16
* 800x480x16 -> 400x240x16 (8kish left for CP heap)

PicoDVI also has support for palettized colors but I wasn't able to get it working. That'd allow you to pick a limited number of rgb888 colors.

> And [the readthedocs documentation](https://docs.circuitpython.org/en/latest/shared-bindings/picodvi/index.html#picodvi.Framebuffer), indicate that the only allowable resolutions are 640x480 and 800x480 and implying that 640x480x16 is an achievable resolution. I am pretty sure this is incorrect.

The key piece of the docs is:

```Monochrome framebuffers (color_depth=1 or 2) will be full resolution. Color framebuffers will be half resolution and pixels will be duplicated to create a signal with the target dimensions.```

todbot, 2023-04-26:
I don't yet care much about making my own Framebuffer, I am content using `board.DISPLAY`, but board.DISPLAY is 320x240x8 when the `board.c` code thinks it's 640x480x8.  And if *I* try to make a 320x240x8 display, it's [an invalid resolution](https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/common-hal/picodvi/Framebuffer.c#L143).

tannewt, 2023-04-26:
Why do you say it "thinks" that?

jposada202020, 2023-04-26:
Agreeing with the comment above. According to my test for the picture in https://github.com/adafruit/circuitpython/issues/7894 is done assuming 320x240 resolution, and you can see the screen is "full"

todbot, 2023-04-26:
Okay, after conversation with @tannewt in Discord, I see the problem.  While you must construct `picodvi.Framebuffer()` with a 640x480 argument or using `board.DISPLAY`,  the resulting display is reported as 320x240 to CircuitPython code, the actual monitor is driven at 640x480 with pixels doubled in both dimensions 

I would still classify this as a bug.   Shouldn't `display.width` and `picodvi.Framebuffer(width=)` have the same value?

As an explicit example using the Feather RP2400 DVI, the following code:

```py
import time
import board, displayio, vectorio

display = board.DISPLAY   # constructed at 640x480 in board.c
print("display width, height:", display.width, display.height)  # prints 320 240

maingroup = displayio.Group()
display.root_group = maingroup
pal = displayio.Palette(2)
pal[0] = 0x00ffff
pal[1] = 0xffff00

wcnt, hcnt = 32, 24  # if we're actually 320x240, this gives us 10 pixel spacing 
for i in range(wcnt):
    for j in range(hcnt):
        x,y = i * 10, j * 10  # 10 pixel spacing
        c = vectorio.Circle(pixel_shader=pal, color_index=(i%2), radius=1, x=x, y=y)
        maingroup.append(c)

while True:
    print("hi")
    time.sleep(1)
```
appears to be working with a 320x240 display, but the monitor is seeing a 640x480 display: 
[image]

dglaude, 2023-04-27:
This going to change (see Scott on Show&Tell): https://www.youtube.com/watch?v=Gcw8rOYaO8U&t=411s

PR #7922 (2023-05-02): Take in framebuffer resolution, not output res
Fixes #7911
commit b08714bb (2023-05-02): Take in framebuffer resolution, not output res

Fixes #7911