adafruit/circuitpython #7912: Distortion with Two Adafruit 64x32 RGB LED Matrix

markbarna, 2023-04-26:
### CircuitPython version

```python
Adafruit CircuitPython 8.3.3 on MatrixPortal M4
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2020 Jeff Epler for Adafruit Industries
#
# SPDX-License-Identifier: MIT

# This example implements a simple two line scroller using
# Adafruit_CircuitPython_Display_Text. Each line has its own color
# and it is possible to modify the example to use other fonts and non-standard
# characters.

import adafruit_display_text.label
import board
import displayio
import framebufferio
import rgbmatrix
import terminalio

bit_depth = 1
base_width = 64
base_height = 32
chain_across = 2
tile_down = 1
serpentine = True

width = base_width * chain_across
height = base_height * tile_down

addr_pins = [board.MTX_ADDRA, board.MTX_ADDRB, board.MTX_ADDRC, board.MTX_ADDRD]
rgb_pins = [
    board.MTX_R1,
    board.MTX_G1,
    board.MTX_B1,
    board.MTX_R2,
    board.MTX_G2,
    board.MTX_B2,
]
clock_pin = board.MTX_CLK
latch_pin = board.MTX_LAT
oe_pin = board.MTX_OE

displayio.release_displays()
matrix = rgbmatrix.RGBMatrix(
                width=width,
                height=height,
                bit_depth=bit_depth,
                rgb_pins=rgb_pins,
                addr_pins=addr_pins,
                clock_pin=clock_pin,
                latch_pin=latch_pin,
                output_enable_pin=oe_pin,
                tile=tile_down, serpentine=serpentine,
            )

# Associate the RGB matrix with a Display so that we can use displayio features
display = framebufferio.FramebufferDisplay(matrix, auto_refresh=False)

# Create two lines of text to scroll. Besides changing the text, you can also
# customize the color and font (using Adafruit_CircuitPython_Bitmap_Font).
# To keep this demo simple, we just used the built-in font.
# The Y coordinates of the two lines were chosen so that they looked good
# but if you change the font you might find that other values work better.
line1 = adafruit_display_text.label.Label(
    terminalio.FONT,
    color=0xff0000,
    text="This scroller is brought to you by CircuitPython RGBMatrix")
line1.x = display.width
line1.y = 8

line2 = adafruit_display_text.label.Label(
    terminalio.FONT,
    color=0x0080ff,
    text="Hello to all CircuitPython contributors worldwide <3")
line2.x = display.width
line2.y = 24

# Put each line of text into a Group, then show that group.
g = displayio.Group()
g.append(line1)
g.append(line2)
display.show(g)

# This function will scoot one label a pixel to the left and send it back to
# the far right if it's gone all the way off screen. This goes in a function
# because we'll do exactly the same thing with line1 and line2 below.
def scroll(line):
    line.x = line.x - 1
    line_width = line.bounding_box[2]
    if line.x < -line_width:
        line.x = display.width

# This function scrolls lines backwards.  Try switching which function is
# called for line2 below!
def reverse_scroll(line):
    line.x = line.x + 1
    line_width = line.bounding_box[2]
    if line.x >= display.width:
        line.x = -line_width

# You can add more effects in this loop. For instance, maybe you want to set the
# color of each label to a different value.
while True:
    scroll(line1)
    scroll(line2)
    #reverse_scroll(line2)
    display.refresh(minimum_frames_per_second=0)
```

### Behavior

I get distortion on the second panel. I have an open thread on the Adafruit support forum and they suggested also opening a bug report here. You can see the thread, which includes images of the distortion, and all of the attempted fixes here: https://forums.adafruit.com/viewtopic.php?t=200635. We have ruled out any issue with the LED panels, themselves.

DJDevon3, 2023-04-29:
I don't have a matrix portal. I have 2 panels, RGB Matrix Featherwing, and M4 Express. Would that be close enough to help beta test this issue?

markbarna, 2023-04-29:
> I don't have a matrix portal. I have 2 panels, RGB Matrix Featherwing, and M4 Express. Would that be close enough to help beta test this issue?

@DJDevon3 , is the code block that I posted unable on your setup? I'm not sure if the hardware is similar enough. That being said, Adafruit is sending me a replacement MatrixPortal so once it arrives, I'll post an update as to whether this issue was due to hardware.

jepler, 2023-04-29:
I have this set-up with matrixportal in the basement, I just need to get it out and update to the latest to find out if I can reproduce a problem:
[image]
with some older circuitpython it did work fine for an extended period of time.

markbarna, 2023-04-29:
@jepler , that's good to know. Would you mind sharing the code and version of CircuitPython that you're using to run that example? I would be interested to try that on mine too.

jepler, 2023-05-02:
OK I've gotten my project out of storage.

I'm using two of the 64x32 LED displays, a matrixportal, and a [4V 2A power supply](https://www.adafruit.com/product/1466)

The power supply reads 5.18V no load and 5.10V while running the code with display on.

I have previously installed a 220µF capacitor on the power input, but I do not know if this is important.

The display looks right with the following versions: 7.3.1 (was previously loaded), 8.0.5, 8.1.0-beta.2

The display initialization code is as follows (I pasted it at the repl for the "hello 7912 image" but it's taken from my working code file)
```
import board, framebufferio, rgbmatrix
matrix = rgbmatrix.RGBMatrix(
    width=128, bit_depth=1,
    rgb_pins=[board.MTX_R1, board.MTX_G1, board.MTX_B1, board.MTX_R2, board.MTX_G2, board.MTX_B2],
    addr_pins=[board.MTX_ADDRA, board.MTX_ADDRB, board.MTX_ADDRC, board.MTX_ADDRD],
    clock_pin=board.MTX_CLK, latch_pin=board.MTX_LAT, output_enable_pin=board.MTX_OE)
DISPLAY = framebufferio.FramebufferDisplay(matrix)
```

Here are some images of my setup:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

markbarna, 2023-05-04:
Thank you for looking at it. I received a replacement MatrixPortal so will
try using that one to run the same code and will post an update.

markbarna, 2023-05-07:
@jepler , apologies for the delay in getting back. I did some further testing with the second MatrixPortal that I received. To your point about a potential power issue, I had been wondering the same thing. Since I had the 2 portals, I decided to try running them both at the same time with each panel using the same power supply but without the data chained together. I'm using Adafruit's 5V/10A power supply with their barrel splitter for just the LED panels and separate USB supplies for each MatrixPortal. One was powered off of my MacBook Pro; the other one was powered off of a 15W USB-C (non-apple) wall jack that I have. Everything looks fine in this configuration with a simple "Hello World!":
[image]

So then I tried chaining them together with the new MatrixPortal. You can see some extra pixels lit around the center line of the "e" on the left panel:
[image]

I tried reversing the panels and it actually looked okay:
[image]

So I tried moving the text all the way over to the left and then again saw some very minor distortion around the center line of the "e":
[image]

And then when I try doing the scroller text, it all falls apart again:
[image]

Given that I'm using that I can power both panels at the same time with the 2 MatrixPortals, I don't imagine that I'm having a power issue. I know that you added a capacitor to your setup. What purpose is that intended to serve? Also, you mentioned that this could be a data integrity issue. Can you elaborate on this a little bit? What would cause that and how would you suggest I investigate? Thanks again for your help on this!

jepler, 2023-05-07:
@PaintYourDragon any thoughts on this kind of distortion? I don't want to get into a technical explanation when I'm not confident of the details, but to me it looks like some kind of problem between the clock & data pixels that is scrambling the data the further left (further from the natrixportal) the pixel is.

PaintYourDragon, 2023-05-08:
Keeping the MatrixPortal and matrices in the same chained order, try just swapping where the power cables go to the panels. Small but nonzero chance of something badly soldered or crimped. I’m highly skeptical this is the problem, but it’s quick and zero cost, worth a try.

Timing is much more likely, and “slop” from the shift registers will accumulate to the left as a chain grows longer. These panels are changing all the time, and what worked on last month’s panels might be different now. Easiest way to experiment with this would be to try the Arduino library (sorry) and test with different overclock settings — 120, 150, 180 MHz — see if any difference. And if so, might need to get in there and change the NOPs for the high and low timing of the pixel clock.

PaintYourDragon, 2023-05-08:
Oh also…if you have a second ribbon cable for chaining the displays, try swapping that out. Small but nonzero chance of something being badly crimped.

dhalbert, 2023-06-21:
@markbarna Did you get any further on this?

markbarna, 2023-06-22:
Hi @dhalbert , unfortunately no. I was a little wary of trying to use an Arduino library to set adjust overclock settings on a MatrixPortal for fear of damaging it. But also I got swamped with work so this is sadly been sitting on my work table waiting for me to have a minute to pick it up again. If you have any other suggestions that might be a little more straightforward, happy to try them. Otherwise, I should hopefully get a chance to dig back in and try messing with the Ardiuno library this summer. Thank you.

RetiredWizard, 2023-06-22:
I don't know if this will be of any possible use as in my case I was only using a single 64x32 Panel but fwiw, when I started filling up my panel with lots of pixels I would see similar distortions as you've displayed. Since it seemed to be related to how many pixels I lit, I guessed that it was a power issue and in my case I was able to boost the voltage on my supply by a volt or so which eliminated the distortion issues. My power supply should have been able to provide more than enough current at the original 5V but I'm thinking there is a voltage drop that gets increased as you light up more pixels (or maybe add more panels) which brings the operating voltage of the pixels below what they need.

RetiredWizard, 2023-06-22:
Scratch that... I just went back and re-tested on my panel and noticed that my power supply had a current limit set for about an amp so when I lit all the pixels up the supply went into current limiting mode. Turning up the voltage reduced the current needed which is why that solved my problem but I just increased the current limit at 5V and that also solved the issue, so increasing the voltage probably isn't they way to go.

RetiredWizard, 2023-06-22:
[image]
[image]

markbarna, 2023-06-24:
@RetiredWizard based on what I'm seeing in the meter displays between the two versions, the distorted one is 1.89W vs 2.17W in the non-distorted version. I'm using this Adafriut https://www.adafruit.com/product/658 5V/10A power supply. Given that it's their power supply, I would have thought it was more than sufficient to drive two of these panels. I know you didn't think your issue was voltage, but I'm wondering if the higher current draw of the 2 panels is causing a voltage drop that is enough to get the distortion given that you're at 5.5V in the non-distorted version.

Given this statement on the LED Matrix page (https://www.adafruit.com/product/2277), I was expecting that this 50W power supply would be sufficient.

> These panels require 13 digital pins (6-bit data, 7-bit control) and a good 5V supply, **up to 4A per panel**. [We suggest our 4A regulated 5V adapter](https://www.adafruit.com/product/1466) and then [connecting a 2.1mm jack](https://www.adafruit.com/products/368). Please check out our tutorial for more details!

RetiredWizard, 2023-06-24:
It would be interesting to measure the voltage at the connection point of your matrix array. Unfortunately, I feel like the precise voltage of a 5 volt power supply can vary by almost a volt and while I haven't done any testing to confirm, I wouldn't be surprised if the output from "regulated" supplies vary under load by up to a volt as well.

PaintYourDragon, 2023-08-05:
Pinging @dhalbert and @jepler here. I was doing some work on the Arduino side of Protomatter today and found that this latest batch of P4 and P5 matrix panels are _super persnickety_ about the clock signal. Today’s 1.5.10 release should fix this, and the change (just moving some NOPs) ought to work in CircuitPython as well.

I think what might’ve happened is occasionally there’s a PR that fixes one person’s flicker issue, unaware that this may lay waste to other situations. So earlier CircuitPython could sometimes behave better than later releases. Testing today on logic analyzer and six different panels, aiming for 50% duty cycle w/20 MHz CLK seems to produce the most reliable output across different hardware, and the code is now configured for such. Added a note to that repo’s README asking folks to open an issue rather than a PR if they have a problem and a “fix” for flicker, as it really needs monitoring and testing across a bunch of parts before merging.

bluelasers, 2023-08-10:
* Remember the fan out! Many panels do not manage the clock line. (These use serial wiring on the clocks. Lowering clock speed will improve this. Use scope to verify performance.)
* Many panels do not use termination as it is likely not required unless the cable is long. 
* Many panels do not manage the upper and lower capacitance, which can create ghosting if the refresh is too high. (Some have MBI5124 which manages the lower. I do not know if they have the RUC7258 which manages upper.)
* The power supply should be cable of enough power to avoid brown out. The power supply must have a switch speed above 1kHz (possibly faster).

bluelasers, 2023-08-10:
Personally I would use 100-400kHz power supply module. (These can be found on Digikey.) Carefully what them with a scope if you get bored. (Use resistive dummy load.)

bluelasers, 2023-08-10:
Personally I would use 100-400kHz power supply module. (These can be found on Digikey.) Carefully watch them with a scope if you get bored. (Use resistive dummy load.)

dhalbert, 2024-01-24:
Closing for now based on PaintYourDragon's testing, etc.

closed 2024-01-24 by dhalbert