adafruit/circuitpython #7917: hit a whoopsi with character lcd

paulrodgers, 2023-04-27:
### CircuitPython version

```python
boot_out:
Adafruit CircuitPython 8.0.5 on 2023-03-31; Adafruit Feather M0 RFM9x with samd21g18
Board ID:feather_m0_rfm9x
UID:09BDE3414E37505120202050052009FF
```

### Code/REPL

```python
import time
import board
import adafruit_character_lcd.character_lcd_i2c as clcd

i2c = board.I2C()  # uses board.SCL and board.SDA
lcd = clcd(i2c, 40, 4)

lcd.message = "Hello, world!"
time.sleep(5)
lcd.clear()
```

### Behavior

get the whoops response in the REPL:
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 8.0.5 on 2023-03-31; Adafruit Feather M0 RFM9x with samd21g18

### Description

tryig to use I2C to driv  char LCD
its older (about 2018)
recycled from the beer-fermenter display we just replaced

### Additional information

Its NOT an adafruit backpack, but its been running on a Rasp Pi for over 4 years
display powers up, backlight works

[lcd_test.zip](https://github.com/adafruit/circuitpython/files/11346090/lcd_test.zip)

dhalbert, 2023-04-27:
The common backpacks use a PCF8574, which is not supported by the Adafruit library. However, it should not normally crash for software reasons.

Are you powering the display and backpack with 5V? If so there may be 5V on the I2C pins from the pull-up resistors. This may cause a hardware issue with the Feather M0.

Do you have a link for the display and backpack, or if not, photos? How is it wired? How was it wired with the RPi, and do you have a link to that software?

paulrodgers, 2023-04-27:
yes, its powered off 5v, e same USB power that runs the feather
I hear you saying that the backpack needs to talk to the feather at 3.3v on the I2C

makes sense, so If I unplug the I2C at the feather end it should no longer crash
will not work, but no crash. but thats not what happens, same result

BTW fantastically fast response, I was expecting tmw morning earliest

if I just get a new adafruit backpack will that run at 3.3v?

dhalbert, 2023-04-27:
This is the latest version of our backpack: https://www.adafruit.com/product/292. The LCD needs 5v for proper contrast, but this new backpack upconverts the 3.3V to 5V. It will probably work with your display, but it's possible it wouldn't. We'd to know more about the display to check on that.

If you disconnect the I2C, you will get an error that there is no I2C connected (it will mention missing pullups).

This appears to be more of a support issue than a CircuitPython bug. I'm going to close this as an issue, but feel free to open a thread in https://forums.adafruit.com/viewforum.php?f=60 and we can continue the discussion there if you'd like. Thanks.

closed 2023-04-27 by dhalbert