adafruit/circuitpython #7919: Improvements to keypad library

regicidalplutophage, 2023-04-28:
Some feature I find missing from keypad library:

- There's no way to use normally closed seitches without external pullups.
In ```keypad.Keys``` if ```pull=True```, the pin is always pulled to the opposite of ```value_when_pressed``` which prevents the use on normally closed switches without external pullups.
Allow the use of normally open switches here and in the ```keypad.KeyMatrix```;
- Add support for SPI bus to be used for ```keypad.ShiftRegisterKeys```;
- Add support for more configurations of shift registers to be used, mainly a shift register matrix with GPIO outputs and shift register inputs;
- Add support for ATSAMD09.

That's all I can think of for now.
Thanks!

s-light, 2023-05-24:
@regicidalplutophage to 1.: so we would need to have `pull=UP` and `pull=DOWN` as option?! 
and a (small) change to line [54 & 55 in Keys.c](https://github.com/adafruit/circuitpython/blob/main/shared-module/keypad/Keys.c#L54)

is it good to extract each request in its own issue?

regicidalplutophage, 2023-05-24:
@s-light 
> so we would need to have `pull=UP` and `pull=DOWN` as option?!

That would the usage of normally closed switches with ```keypad.Keys```.
For matrices ```pressed``` and ```released``` would have to be inverted and same for shift register.

> is it good to extract each request in its own issue?

I don't have an opinion on that.

dhalbert, 2023-05-24:
Please do split issues up, since they can be fixed one at a time.

> so we would need to have `pull=UP` and `pull=DOWN` as option?!

Instead of that, I think I would add a `normally_open=True` optional argument, to handle normally closed switches.

I'm not sure we want to add lots more features to `keypad`, because it makes it larger and it might stop fitting on smaller builds. But adding some kind of access to the `Event` queues would make it possible to do some of this in Python instead of in C.

regicidalplutophage, 2023-05-24:
Before I start opening new issues, let's think if one big change can solve a bunch of problems at once.
Then I think keypad should at least take ```digitalio.DigitalInOut```, ```adafruit_tca8418.DigitalInOut```, ``` adafruit_74hc595.DigitalInOut``` and the like.
This way we can remove shift register scanner from keypad library altogether without losing shift register support and still gain support for shift register or mixed matrices.

kolkmvd, 2023-06-30:
_Add support for more configurations of shift registers to be used, mainly a shift register matrix with GPIO outputs and shift register inputs;_

How do you mean this? I have a board where I have 4 HC165 in parallel, sharing the same CLOCK and LATCH, but each on a different DATA. To make this supported I was thinking about adding a function **ShiftRegister.addDataPin(data: microcontroller.Pin, keycount: int)** that would resize scanarrays and increase the key_count after the initial set-up. 

Would this interfere with you suggestion?

kolkmvd, 2023-07-07:
Created #8143, which allows the data and key_count arguments to be a either a single pin as currently, but also a list as well.

s-light, 2023-08-22:
i created two feature-requests out of these comments here..
i think - as @regicidalplutophage already formulated - especially #8316 would solve a bunch of things..
also half of #8315