adafruit/circuitpython #7923: ESP32-S3 TFT could not work with the HTU31D sensor

jposada202020, 2023-04-29:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.2 on 2023-04-26; Adafruit Feather ESP32-S3 TFT with ESP32S3
```

### Code/REPL

```python
import time
import board
import adafruit_htu31d

i2c = board.I2C()  # uses board.SCL and board.SDA
htu = adafruit_htu31d.HTU31D(i2c)

while True:

    temperature, relative_humidity = htu.measurements
    print("Temperature: %0.1f C" % temperature)
    print("Humidity: %0.1f %%" % relative_humidity)
    print("")

    time.sleep(1)
```

### Behavior

Adafruit CircuitPython 8.1.0-beta.2 on 2023-04-26; Adafruit Feather ESP32-S3 TFT with ESP32S3
>>> import ht
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ht.py", line 10, in <module>
  File "/lib/adafruit_htu31d.py", line 179, in measurements
RuntimeError: Invalid CRC calculated

### Description

I have removed the raise statement from the library and both the temperature and humidity checksum are wrong.. I have tested the same ESP32-S3 with an TMP117 sensor and works without any issue.
This was reported from a user in discord, only difference is that the ESP32-S3 is a reverse TFT.

### Additional information

I have tested the same sensor with a 
```python
Adafruit CircuitPython 8.1.0-beta.2 on 2023-04-26; Adafruit Feather RP2040 USB Host with rp2040

```

### Result

```python
Adafruit CircuitPython 8.1.0-beta.2 on 2023-04-26; Adafruit Feather RP2040 USB Host with rp2040
>>> import ht
Temperature: 26.8 C
Humidity: 42.6 %

Temperature: 26.8 C
Humidity: 42.5 %

Temperature: 26.7 C
Humidity: 42.5 %

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ht.py", line 19, in <module>
KeyboardInterrupt: 
```

dhalbert, 2023-05-01:
@jposada202020 Do you have a similar ESP32-S2 you could test it with? It might be an I2C clock-stretching or similar issue, which the S3 can have trouble with. Thanks.

jposada202020, 2023-05-01:
@dhalbert works fine with the S2 :)

```python
Adafruit CircuitPython 8.1.0-beta.2 on 2023-04-26; Adafruit Feather ESP32-S2 TFT with ESP32S2
>>> import t2
Temperature: 21.0 C
Humidity: 54.7 %

Temperature: 20.9 C
Humidity: 54.7 %

Temperature: 20.9 C
Humidity: 54.7 %

Temperature: 21.0 C
Humidity: 54.8 %

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "t2.py", line 15, in <module>
KeyboardInterrupt: 

```

dhalbert, 2023-06-21:
Retest after updating to ESP-IDF 5.

mikeysklar, 2023-08-07:
Slowing down I2C bus to 70 kHz got my ESP32-S3 TFT (REV) working with the HTU31D.

`i2c = busio.I2C(board.SCL, board.SDA, frequency=70_000)
`

https://github.com/adafruit/Adafruit_CircuitPython_HTU31D/issues/12

dhalbert, 2023-11-17:
Tested on CircuitPython 9.0.0-alpha.5, which was updated to ESP-IDF 5.1. There are many I2C fixes. The test program now works nicely. Closing as fixed in 9.0.0 and unfixable in 8.x.x.

closed 2023-11-17 by dhalbert

mikeysklar, 2023-11-17:
Thank you @dhalbert.