adafruit/circuitpython #7926: CircuitPython HARD FAULTs crashes hard after several hours from async/await loops

chilipeppr, 2023-05-02:
### CircuitPython version

```python
When running this simple code that creates several async/await tasks that run in parallel, CircuitPython will HARD_FAULT randomly roughly every several hours.

I have tested this on CircuitPython 8.0.5 and 8.1.0 and they both act similarly.
```

### Code/REPL

```python
# main_test_why_circuitpython_hardfaults.py
# This file will just run a simple asyncio infinite loop to see if that causes
# a hard fault after 24 hours.

import asyncio
import time
from adafruit_datetime import datetime as adatetime

class Dashboard:
    def __init__(self):

        print("Initting Dashboard")

        # Create several infinite loop tasks that loop each 0.01 to 0.5 seconds
        self.loop_task1 = asyncio.create_task(self.asyncTaskLoop("task1", 100, 0.1))
        self.loop_task2 = asyncio.create_task(self.asyncTaskLoop("task2", 50, 0.2))
        self.loop_task3 = asyncio.create_task(self.asyncTaskLoop("task3", 100, 0.1))
        self.loop_task4 = asyncio.create_task(self.asyncTaskLoop("task4", 20, 0.5))
        self.loop_task5 = asyncio.create_task(self.asyncTaskLoop("task5", 17, 0.6))
        self.loop_task6 = asyncio.create_task(self.asyncTaskLoop("task6", 100, 0.1))
        self.loop_task7 = asyncio.create_task(self.asyncTaskLoop("task7", 50, 0.2))
        self.loop_task8 = asyncio.create_task(self.asyncTaskLoop("task8", 100, 0.1))
        self.loop_task9 = asyncio.create_task(self.asyncTaskLoop("task9", 20, 0.5))
        self.loop_task10 = asyncio.create_task(self.asyncTaskLoop("task10", 17, 0.6))

    async def asyncTaskLoop(self, taskName, howOftenTellUsYouLooped, loopDelay):
        print("{}: Starting infinite loop".format(taskName))
        
        ctr = 0
        startTime = time.monotonic()

        while True:
            await asyncio.sleep(loopDelay) # don't forget the await

            ctr += 1
            if ctr > howOftenTellUsYouLooped:
                
                print("{} {}: Looped {} times waiting {}s on each loop. Should have taken {}s, but took {}s".format(
                    adatetime.now(), taskName, howOftenTellUsYouLooped, loopDelay, 
                    loopDelay*howOftenTellUsYouLooped, 
                    time.monotonic() - startTime))
                ctr = 0
                startTime = time.monotonic()
        

async def main():
    
    d = Dashboard()

    await asyncio.gather(
        d.loop_task1, d.loop_task2, d.loop_task3, d.loop_task4, d.loop_task5,
        d.loop_task6, d.loop_task7, d.loop_task8, d.loop_task9, d.loop_task10
    )  # Don't forget the await!

asyncio.run(main())
```

### Behavior

[... 236 lines cut ...]
# The user gets to toggle the kill switch and if it's off, the main code
# runs, but if it's on then the main.py code just exits so you can debug
#
# We are literally just doing one test here on boot. So this is very
# straightforward code.

import board
import digitalio

class KillSwitch:

    def __init__(self):

        print("Initting Kill Switch library...")
        self._pinKillSwitchPin = board.IO1

        self._pinKillSwitch = digitalio.DigitalInOut(self._pinKillSwitchPin)
        self._pinKillSwitch.switch_to_input(pull=digitalio.Pull.UP)

    def isKillSwitchOn(self):

        val = not self._pinKillSwitch.value
        print("KillSwitch:", val)
        return val    
```

anecdata, 2023-05-02:
Wemos S2 Mini?

chilipeppr, 2023-05-02:
Yes.

RetiredWizard, 2023-05-03:
I increased the number of asyncio tasks to 30 and started your test program on a UM Feather S2 and an Adafruit Feather ESP32-S2 about 13 hours ago. They haven't crashed yet but I'll let them run over night. I'm assuming that adding the extra tasks didn't mess the test up. Unfortunately, I don't have a Wemos S2 Mini to test on, hopefully it's not hardware or port specific. 

I did notice that the Lolin S2 boards are among the relatively few S2 boards that have CIRCUITPY_ESP_FLASH_FREQ set to 80m in mpconfigboard.mk. Your test script doesn't access flash so that's probably not relevant though.

chilipeppr, 2023-05-03:
I was actually using a Lolin S2 Mini, not a Wemos S2 Mini, but aren't they basically the same thing?

I've been running this even further since I posted and I did get a HARD_FAULT again a couple hours ago. Here's my log below. I also seem to sometimes get the WATCHDOG as well as a reason for a reboot. So my period between HARD_FAULTs is about 22 hours.

I'm building a Marble Run for the local school's STEM lab to inspire the kids and this thing needs to run 24x7 so the kids can hit the button whenever they want all day to launch the marbles down a huge track. So was noticing the hard faults on an ongoing basis and initially thought it was my code, but it really seems like it's just the OS doing it.

```
From safemode.py. Kill switch off. 2023-05-01 21:45:06, supervisor.RunReason.STARTUP, microcontroller.ResetReason.SOFTWARE, supervisor.SafeModeReason.HARD_FAULT
From main.py. Kill switch off. 2023-05-01 21:44:54, supervisor.RunReason.STARTUP, microcontroller.ResetReason.SOFTWARE, supervisor.SafeModeReason.NONE
From safemode.py. Kill switch off. 2023-05-02 00:18:08, supervisor.RunReason.STARTUP, microcontroller.ResetReason.SOFTWARE, supervisor.SafeModeReason.HARD_FAULT
From main.py. Kill switch off. 2023-05-02 00:18:03, supervisor.RunReason.STARTUP, microcontroller.ResetReason.SOFTWARE, supervisor.SafeModeReason.NONE
From main.py. Kill switch off. 2023-05-02 06:07:34, supervisor.RunReason.AUTO_RELOAD, microcontroller.ResetReason.SOFTWARE, supervisor.SafeModeReason.NONE
From main.py. Kill switch off. 2023-05-02 06:20:17, supervisor.RunReason.REPL_RELOAD, microcontroller.ResetReason.SOFTWARE, supervisor.SafeModeReason.NONE
From safemode.py. Kill switch off. 2023-05-02 11:00:43, supervisor.RunReason.STARTUP, microcontroller.ResetReason.WATCHDOG, supervisor.SafeModeReason.WATCHDOG
From main.py. Kill switch off. 2023-05-02 10:59:46, supervisor.RunReason.STARTUP, microcontroller.ResetReason.SOFTWARE, supervisor.SafeModeReason.NONE
From safemode.py. Kill switch off. 2023-05-02 18:22:22, supervisor.RunReason.STARTUP, microcontroller.ResetReason.SOFTWARE, supervisor.SafeModeReason.HARD_FAULT
From main.py. Kill switch off. 2023-05-02 18:23:10, supervisor.RunReason.STARTUP, microcontroller.ResetReason.SOFTWARE, supervisor.SafeModeReason.NONE
```

RetiredWizard, 2023-05-03:
Yea, I think they are the same board. Hopefully I'll see a crash by morning, If not, maybe I'll try loading the Lolin firmware on one of them and see if I can reproduce the crash that way.

chilipeppr, 2023-05-03:
Here's my bout_out.txt, but keep in mind I saw the same thing on 8.0.5.

```
Adafruit CircuitPython 8.1.0-beta.1 on 2023-03-30; S2Mini with ESP32S2-S2FN4R2
Board ID:lolin_s2_mini
UID:487F307D7D25
```

RetiredWizard, 2023-05-04:
I had a bit of a glitch overnight last night which killed my terminal sessions, so from yesterday's test runs all I know is that neither board crashed after about 13 hours.

I started the tests up again this morning but this time, I loaded the Lolin S2 mini firmware up on one of the boards first. The test program has been running again for about 13 hours. I'll check them again in the morning and hopefully at least one of them will have hard faulted.

chilipeppr, 2023-05-04:
Sounds good. Yeah, I left mine running today again too, but I actually had
accidentally left it in the REPL, so missed out on my test today as well.

chilipeppr, 2023-05-04:
Had a HARD_FAULT again last night. 

The timestamp is odd. Safemode got hit at 3:33AM. Then it rebooted into normal mode which should be about 10 seconds later, but the clock says 3:27AM. I do the RTC lookup when booting in normal mode, so it just makes me think the clock runs fast on the ESP32. Is it possible a fast clock throws stuff off over time and that's what causes a hard fault?

```
From safemode.py. Kill switch off. 2023-05-04 03:33:58, supervisor.RunReason.STARTUP, microcontroller.ResetReason.SOFTWARE, supervisor.SafeModeReason.HARD_FAULT
From main.py. Kill switch off. 2023-05-04 03:27:43, supervisor.RunReason.STARTUP, microcontroller.ResetReason.SOFTWARE, supervisor.SafeModeReason.NONE
```

chilipeppr, 2023-05-04:
> I have tested this on 2 different Lolin S2 Mini's and same issue, but I
> could try this on other ESP32's now that you mention it could be specific
> to the version of the chip they used.

Do you think it has anything to do with me activating the display?

RetiredWizard, 2023-05-04:
Well I'm not doing anything with the display, I'd suggest you run the same test you have posted above. I don't think just having the display physically attached should be an issue if the software doesn't address it.

Another thought, how are you powering the board?

chilipeppr, 2023-05-04:
I do have the display regurgitating the standard output, so code would be
executing in those display classes. I just commented out that part of the
code so I'm ONLY testing the async tasks. If this doesn't crash on me then
it would have to be the display causing this. If it does crash, one other
idea is that I am using the Web Workflow to see the serial output. Perhaps
it's the Wifi classes causing it.

As for powering the board, I'm just using a normal USB-C wall wart for a
Raspberry Pi that's powering the ESP32-S2, so plenty of amps coming out of
that power supply as it's a 3.5A one I had lying around.

I'll let you know what I see over the next 24 hours.

RetiredWizard, 2023-05-04:
I've switched my monitoring to the web workflow serial terminal as well....

RetiredWizard, 2023-05-05:
The UM Feather S3 has now been running 52 hours without crashing. Somewhere between 25 and 52 hours the Adafruit Feather S2 with the Lolin S2 Mini firmware crashed but I didn't capture any information as a terminal wasn't connected at the time. Before I build a debug image and try and capture a traceback, maybe I'll try a Lolin build with CIRCUITPY_ESP_FLASH_FREQ set to 40m.

chilipeppr, 2023-05-05:
I had another crash last night. I did not run any display code. So, this
was a clean run of just the asyncio.

RetiredWizard, 2023-05-08:
The UM Feather S2 has now been running for 6 days without crashing. I restarted the Adafruit Feather S2 running the Lolin firmware 3 days ago and it hasn't crashed either.

The only time I've seen a crash is when the terminal has been disconnected so I decided to use your logging routine and see if I could recreate the crashes without the serial terminal session. It turns out your code won't run under 8.0.5 because the safemode.py feature isn't implemented in the 8.0.5 line yet.

I'll rebuild my tests using the 8.1.0 build and see if I can reproduce the issue there.

chilipeppr, 2023-05-08:
Ok, that's promising to me actually. It means there is not something deep
in the bowels of CircuitPython or ESP-IDF causing this. Maybe I should just
change my board and then I won't have crashes anymore. Or perhaps I'll try
to slow down that frequency on the Flash chip like you were commenting on a
while ago.

RetiredWizard, 2023-05-12:
I've been running my two S2 boards for another couple days without any luck reproducing the issue.

The Lolin board I ordered came in, unfortunately, it turned out to be one of the likely [counterfeit](https://forums.adafruit.com/viewtopic.php?t=197737) devices based on the silk screen and lack of PS RAM. I was able to build a custom CircuitPython binary which started the REPL, however I couldn't run the test script for more than 15 to 20 minutes and the REPL/Web Access interface would periodically hang for short periods of time.

I ordered the board through Walmart so it should be easy to return but I'm not sure it's worth re-ordering another board directly from China as it will take over a month to get here.