adafruit/circuitpython #7937: Setting CPU frequency on RP2040

TimTrudeau, 2023-05-05:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.2 on 2023-02-14; Waveshare RP2040-Zero with rp2040
```

### Code/REPL

```python
import microcontroller
if __name__ == '__main__':
    print(microcontroller.cpus[0].frequency)
    print(microcontroller.cpus[1].frequency)
    microcontroller.cpu.frequency = 1000000
    print(microcontroller.cpu.frequency)
```

### Behavior

125000000
125000000
Traceback (most recent call last):
  File "stdin", line 5, in module
AttributeError: can't set attribute 'frequency'

### Description

Documentation states this should work on the RP2040
frequency: [int](https://docs.python.org/3/library/functions.html#int)[](https://docs.circuitpython.org/en/latest/shared-bindings/microcontroller/index.html#microcontroller.Processor.frequency)

    The CPU operating frequency in Hertz.

    Limitations: On most boards, frequency is read-only. Setting the frequency is possible on RP2040 boards and some i.MX boards.

Neradoc, 2023-05-05:
Hi, it was just added, it's only available in current `main`.
Which you get from the board's page https://circuitpython.org/board/waveshare_rp2040_zero/
Going through the "Browse S3" link, choosing the language, and downloading the top UF2.

You can consult the latest stable build documentation with the panel on the bottom left.
https://docs.circuitpython.org/en/8.0.x/shared-bindings/microcontroller/index.html#microcontroller.Processor.frequency

closed 2023-05-05 by Neradoc

TimTrudeau, 2023-05-05:
I installed 
Adafruit CircuitPython 8.0.5 on 2023-03-31; Waveshare RP2040-Zero with rp2040
microcontroller.cpu.frequency = 1000000
still generates AttributeError: can't set attribute 'frequency'

bill88t, 2023-05-05:
8.0.5 doesn't have it. Only the latest builds have this functionallity.
You can get such a build using the instructions Neradoc provided in the comment above.

Alternatively you can wait for the stable 8.1.0 release.

TimTrudeau, 2023-05-05:
I figured that out. Pulled the beta and now it works. Thanks!