adafruit/circuitpython #7940: ping too frequently results in resource exhaustion, exception

jepler, 2023-05-07:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.2-24-gd2aca7eba0-dirty on 2023-05-05; ESP32-S3-DevKitC-1-N8R2 with ESP32S3
Board ID:espressif_esp32s3_devkitc_1_n8r2
UID:866B3BD27A8F
```

### Code/REPL

```python
import time
import wifi
from secrets import secrets

DELAY = 0.5  # this is fine, but lower values lead to exceptions after a few pings

wifi.radio.connect(secrets['ssid'], secrets['password'])
while True:
    print(f"LAN ping: {wifi.radio.ping(wifi.radio.ipv4_gateway)}s")
    time.sleep(DELAY)
```

### Behavior

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
espidf.IDFError: Generic Failure
```

### Description

The esp-idf implementation of ping creates a task which actually sends and receives the ping packets.

When `esp_ping_delete_session` is called, the underlying resources are not immediately freed. Instead, this waits until a (hard coded) 1 second delay expires in the ping task, which then checks a flag, frees resources, and terminates itself.

### Additional information

Before #7938 this led to a hard crash and entry into safe mode(#5980), as the return value of `esp_ping_new_session` wasn't checked. Now that the return value is checked, there's a (recoverable, potentially) exception. That's an improvement on the status quo ante but it'd be nice if ping did not have this resource exhaustion problem. This likely means creating our own ping implementation, rather than using esp-idf's.

jepler, 2023-06-21:
Once we've updated esp-idf in CP 9 we should revisit this and see whether anything's improved. If not, we have options from a "1s sleep workaround" to implementing our own ping using LWIP API calls.

jepler, 2023-12-06:
This is unchanged in current idf. If the current state of affairs is unsatisfactory, we should probably write our own ping out of lwip socket calls.

tannewt, 2024-01-31:
I don't think this is urgent for 9.0.

dhalbert, 2024-06-25:
Partially fixed by #9350, which introduces a delay as necessary to prevent errors, but ideally we'd redo ping so it doesn't have delays or errors at all.

gitcnd, 2024-06-26:
I notice when testing on Adafruit CircuitPython 9.1.0-beta.1-24-g24f2e4f994-dirty on 2024-06-23; Ai Thinker ESP32-CAM noSPI with ESP32

that ping "times out" in just 500ms

that's too short.

```python
def ping(shell, cmdenv):
    if len(cmdenv['args']) < 2:
        print("usage: ping <address>")
        return

    dom = cmdenv['args'][1]

    pool = socketpool.SocketPool(wifi.radio)

    try:
        addr_info = pool.getaddrinfo(dom, 80)  # Using port 80 for HTTP
        ip = addr_info[0][4][0]
        ip1 = ipaddress.ip_address(ip)
    except Exception as e:
        print(f'Error getting IP address: {e}')
        return

    print(f"PING {dom} ({ip}) 56(84) bytes of data.")

    packet_count = 4
    transmitted = 0
    received = 0
    total_time = 0
    times = []

    for seq in range(1, packet_count + 1):
        transmitted += 1
        start_time = time.monotonic()

        result = wifi.radio.ping(ip1)
        rtt = (time.monotonic() - start_time) * 1000  # Convert to milliseconds
        total_time += rtt

        if result is not None:
            received += 1
            times.append(rtt)
            print(f"64 bytes from {ip}: icmp_seq={seq} time={rtt:.1f} ms")
        else:
            print(f"Request timeout for icmp_seq {seq}")

        if rtt<1000 and seq<packet_count:
            time.sleep((1000-rtt)/1000)

    print(f"--- {ip} ping statistics ---")
    print(f"{transmitted} packets transmitted, {received} received, {((transmitted - received) / transmitted) * 100:.0f}% packet loss, time {total_time:.0f}ms")

    if times:
        min_time = min(times)
        avg_time = sum(times) / len(times)
        max_time = max(times)
        mdev_time = (sum((x - avg_time) ** 2 for x in times) / len(times)) ** 0.5
        print(f"rtt min/avg/max/mdev = {min_time:.3f}/{avg_time:.3f}/{max_time:.3f}/{mdev_time:.3f} ms")

```

[image]

(4 pings took 2000ms = 500ms/ping)

dhalbert, 2024-06-26:
The ping functionality provided by ESP-iDF sets up a task that runs for a minimum of 1-2 seconds. That's hard-wired. That's the fundamental issue here. We could reimplement that "by hand", but we haven't done that yet.

gitcnd, 2024-06-27:
Is there some kind of internal cpy "call" that can be made to poll/process interruptions (like ^C)?

Maybe the answer is to sit in a loop doing that until the reply or a timeout happens?

I notice the ping doco:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/icmp_echo.html
lets us adjust anything we want, but CPY is just doing 
      ```esp_ping_config_t ping_config = ESP_PING_DEFAULT_CONFIG();```

Which is this:-
```
#define ESP_PING_DEFAULT_CONFIG() { \
    .count = 4,                     \
    .interval_ms = 1000,            \
    .timeout_ms = 1000,             \
    .data_size = 32,                \
    .ttl = 64,                      \
    .target_addr = NULL,            \
    .task_stack_size = 2048,        \
    .task_prio = 5,                 \
    .interface = ESP_IF_WIFI_STA    \
}
```

It might be nice to not hardcode all those defaults like that, and let us users actually have access to the things we want to use?

I also wonder if "wifi/Radio.c" is the right place?  What about ethernet (wired) users?  Disclaimer - I've not looked at cpy's source (other than porting boards) so I have very little clue about whether that's relevant/important (maybe it already works fine)

common-hal/wifi/Radio.c

```C++
mp_int_t common_hal_wifi_radio_ping(wifi_radio_obj_t *self, mp_obj_t ip_address, mp_float_t timeout) {
    esp_ping_config_t ping_config = ESP_PING_DEFAULT_CONFIG();
    ipaddress_ipaddress_to_esp_idf(ip_address, &ping_config.target_addr);
    ping_config.count = 1;

    size_t timeout_ms = timeout * 1000;

    esp_ping_handle_t ping;
    CHECK_ESP_RESULT(esp_ping_new_session(&ping_config, NULL, &ping));
    esp_ping_start(ping);

    uint32_t received = 0;
    uint32_t total_time_ms = 0;
    uint32_t start_time = common_hal_time_monotonic_ms();
    while (received == 0 && (common_hal_time_monotonic_ms() - start_time < timeout_ms) && !mp_hal_is_interrupted()) {
        RUN_BACKGROUND_TASKS;
        esp_ping_get_profile(ping, ESP_PING_PROF_DURATION, &total_time_ms, sizeof(total_time_ms));
        esp_ping_get_profile(ping, ESP_PING_PROF_REPLY, &received, sizeof(received));
    }
    uint32_t elapsed_time = 0xffffffff;
    if (received > 0) {
        esp_ping_get_profile(ping, ESP_PING_PROF_TIMEGAP, &elapsed_time, sizeof(elapsed_time));
    }
    esp_ping_delete_session(ping);

    return elapsed_time;
}
```

There's also a wealth of data returned in the ping callback SEQNO and TTL are very important) which many of us would love to have as well - it's a shame that all seems to be getting discarded?

```C++
static void test_on_ping_success(esp_ping_handle_t hdl, void *args)
{
    // optionally, get callback arguments
    // const char* str = (const char*) args;
    // printf("%s\r\n", str); // "foo"
    uint8_t ttl;
    uint16_t seqno;
    uint32_t elapsed_time, recv_len;
    ip_addr_t target_addr;
    esp_ping_get_profile(hdl, ESP_PING_PROF_SEQNO, &seqno, sizeof(seqno));
    esp_ping_get_profile(hdl, ESP_PING_PROF_TTL, &ttl, sizeof(ttl));
    esp_ping_get_profile(hdl, ESP_PING_PROF_IPADDR, &target_addr, sizeof(target_addr));
    esp_ping_get_profile(hdl, ESP_PING_PROF_SIZE, &recv_len, sizeof(recv_len));
    esp_ping_get_profile(hdl, ESP_PING_PROF_TIMEGAP, &elapsed_time, sizeof(elapsed_time));
    printf("%d bytes from %s icmp_seq=%d ttl=%d time=%d ms\n",
           recv_len, inet_ntoa(target_addr.u_addr.ip4), seqno, ttl, elapsed_time);
}
```

The "good" news is that what "ping" actually returns seems undocumented? ( https://docs.circuitpython.org/en/latest/shared-bindings/wifi/index.html ) so giving back all the data we got should be easy to implement without breaking anything.

gitcnd, 2024-06-27:
If anyone wants me to dive in and take a stab at fixing this - let me know (and if you've got a spare minute - fill me in on anything relevant I'd need to keep in mind to do that - I have no idea of the related tech considerations, nor the process/politics of "core" PRs etc - and I'd prefer not to put effort into submissions that get thrown away, like nearly all of the micropython contributions do https://github.com/micropython/micropython/pulls?q=is%3Apr+is%3Aunmerged ).
FYI - I'm a professional code porter, with about 35 years experience

jepler, 2024-06-28:
I don't _think_ any of the settings in `esp_ping_config_t` control this timeout. Rather, it's hard-coded as 1000ms in esp-idf. timeout_ms and PING_CHECK_START_TIMEOUT_MS seem to be distinct things despite the names both being about "timeout".

Both our current wifi ports use lwip. One possibility would be to factor out the ping implementation in `raspberrypi/common-hal/wifi/Wifi.c` and then use it in both ports. It uses a simple blocking loop with timeout & keyboard interrupt checking.

Another alternative is to offer a change to esp-idf so that a ping task can be canceled immediately. I don't know exactly what this would look like, but I imagine that the task cancellation could in some way wake up the task immediately instead of waiting for that 1-second timeout. We use a fork of esp-idf so we could take a PR there to improve things, especially if it looks on track to be included in a future esp-idf.

---

Just on the topic of micropython PRs showing as "closed unmerged": the micropython maintainers use a workflow in which PRs are usually NOT merged by clicking a button in the github web ui. Instead, maintainers apply the changes locally and push them up. Github is insufficiently flexible/aware of alternative workflows that it ends up thinking the PRs were not merged. A principal author of curl recently wrote about this Github problem: https://daniel.haxx.se/blog/2024/06/11/why-curl-closes-prs-on-github/ -- typical micropython PR of mine that _was_ incorporated but shows as "closed unmerged": https://github.com/micropython/micropython/pull/3802