adafruit/circuitpython #7943: the internal `errno` module is undocumented

jepler, 2023-05-07:
it would be nice if it were documented.

jepler, 2023-05-07:
It's actually documented at docs/library/errno.html but this is not obvious...

dhalbert, 2023-05-16:
@tekktrik Yes, that makes sense. The shared-bindings list is generated by a script, right? Note that `errno` and other "library" modules are already accessible in the Module Support Matrix, so this might be a simple change to the script.

tekktrik, 2023-05-16:
Sounds good!  I'll take this and make the relevant change!

tekktrik, 2023-05-17:
It looks like this one (and others in this section) are could be moved over to the Core Modules section, but I'm not sure I could put it in order alphabetically.  Would it be work combining the "MicroPython libraries" section entirely into the "Core Modules" section?

tekktrik, 2023-05-17:
https://github.com/adafruit/circuitpython/blob/main/shared-bindings/index.rst

jepler, 2023-06-21:
@tekktrik are you planning to work on this? Otherwise, we'll just move it to long term.

tekktrik, 2023-06-21:
I'm looking for clarification on my question above, but otherwise happy to complete it!

dhalbert, 2023-06-21:
> It looks like this one (and others in this section) are could be moved over to the Core Modules section, but I'm not sure I could put it in order alphabetically. Would it be work combining the "MicroPython libraries" section entirely into the "Core Modules" section?

I think this would be OK, perhaps adding a warning at the top that though they appear to be in alphabetical order, some modules are at the bottom. Note that `help()` is already at the bottom. Could also add a link to the Module Support Matrix with a note that says that's another way in to the list.

We _could_ also list everything manually, but then every new module has to be added by hand, or else we have to automate that part.

There is also some way to add a hook routine that gets called that can re-sort. This may or may not help: https://stackoverflow.com/questions/8324959/how-can-i-reverse-order-the-toctree-provided-by-the-glob-flag-option

Note that alphabetization by title instead of filename has been proposed but not implemented:https://github.com/sphinx-doc/sphinx/issues/1430.

dhalbert, 2025-03-22:
Fixed by #10160.

PR #10160 (2025-03-21): Fix multiple documentation issues.
- Fixes #10037.
- Fixes #10147.
- Fixes #9690.
- Fixes #9733.
- Fixes #7943.
- Fixes #7743.
- Fixes #4493.

Also fixes a few minor typos I encountered elsewhere.
closed 2025-03-22 by dhalbert