adafruit/circuitpython #7944: metro m7 no longer gets midi events

jepler, 2023-05-08:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.2 on 2023-04-26; Metro MIMXRT1011 with IMXRT1011DAE5A
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import time

import usb_midi

import adafruit_midi

# from adafruit_midi.channel_pressure        import ChannelPressure
from adafruit_midi.control_change import ControlChange
from adafruit_midi.note_off import NoteOff
from adafruit_midi.note_on import NoteOn
from adafruit_midi.pitch_bend import PitchBend

# A subset of messages/events
# pylint: disable=unused-import
from adafruit_midi.timing_clock import TimingClock

# 0 is MIDI channel 1
midi = adafruit_midi.MIDI(midi_in=usb_midi.ports[0], in_channel=0)

print("Midi input test")

# Convert channel numbers at the presentation layer to the ones musicians use
print("Input channel:", midi.in_channel + 1)

while True:
        msg = midi.receive()
        if (msg := midi.receive()) is not None:
            print(time.monotonic(), msg)
```

### Behavior

no midi events are ever printed

### Description

Under 8.0.5, the test program prints note on/off messages when I send MIDI events to it using a virtual keyboard on my PC.

Under 8.1.0-beta.2, the notes are not received

### Additional information

8.1.0-beta.0 tests OK, 8.1.0-beta.1 fails my test. So the problem seems to have been introduced between those two versions.

@jedgarpark saw this too while working on preparing a demo last wednesday, I only now had time to confirm it.

jepler, 2023-05-08:
@tannewt I did a "git bisect" and found that it was your PR that introduced the problem. I haven't done any further troubleshooting beyond this. I mention it in case you want to step in or have any info to provide...
```
5bb8a7a7c68781e6edbd3aebaf7eeeecb310bd86 is the first bad commit
commit 5bb8a7a7c68781e6edbd3aebaf7eeeecb310bd86
Author: Scott Shawcroft <scott@tannewt.org>
Date:   Tue Feb 28 15:07:35 2023 -0800

    Improve iMX RT performance
```
<details>
<summary>
my bisect log</summary>

```git bisect start
# status: waiting for both good and bad commits
# bad: [b07de078f9dd640c28668cad3b5a5728a1d1024e] Merge pull request #7818 from CDarius/m5stack_stickc_fixes
git bisect bad b07de078f9dd640c28668cad3b5a5728a1d1024e
# status: waiting for good commit(s), bad commit known
# good: [cb6201d952c7be39024add8d91b79803ee191485] Merge pull request #7659 from dhalbert/gifio-displayio-doc
git bisect good cb6201d952c7be39024add8d91b79803ee191485
# bad: [461d833c1c9ead38542dbbfe665af5c3446035bd] Merge pull request #7763 from dhalbert/update-tinyusb-remove-usb_descriptor
git bisect bad 461d833c1c9ead38542dbbfe665af5c3446035bd
# good: [bca7d6e99933ef77af7cb29e5cb8c25d735051e1] Merge pull request #7717 from jepler/warn-too-new-uncrustify
git bisect good bca7d6e99933ef77af7cb29e5cb8c25d735051e1
# good: [22d7c0b8e904b5324155dbd668d41a8840b1d044] Merge pull request #7736 from bwshockley/fig-pi-update
git bisect good 22d7c0b8e904b5324155dbd668d41a8840b1d044
# good: [32a596df39ab8600cbc3d5f4ca4e0c65e5551033] Merge pull request #7757 from adafruit/8.0.x
git bisect good 32a596df39ab8600cbc3d5f4ca4e0c65e5551033
# good: [c78502d8a589a356e6eeb398762c62b910fc257f] Merge pull request #7746 from FoamyGuy/diskinfo_api
git bisect good c78502d8a589a356e6eeb398762c62b910fc257f
# bad: [7ee85408a1db9b79691692c74fa7ae10a0cf05b0] Merge pull request #7755 from isacben/block-device-arguments-typing
git bisect bad 7ee85408a1db9b79691692c74fa7ae10a0cf05b0
# bad: [67e0a49a1f29e87c4d7b35331dbb6c751b37ee49] Merge pull request #7724 from tannewt/get_perfbench_running
git bisect bad 67e0a49a1f29e87c4d7b35331dbb6c751b37ee49
# bad: [bdf592089ae461434a03ba0f185dfce3d09953fb] Fix .bin, .hex and .uf2 with new linker sections
git bisect bad bdf592089ae461434a03ba0f185dfce3d09953fb
# bad: [5bb8a7a7c68781e6edbd3aebaf7eeeecb310bd86] Improve iMX RT performance
git bisect bad 5bb8a7a7c68781e6edbd3aebaf7eeeecb310bd86
# first bad commit: [5bb8a7a7c68781e6edbd3aebaf7eeeecb310bd86] Improve iMX RT performance
```

</details>

jepler, 2023-05-08:
when going across that pair of commits, the CIRCUITPY drive gets resized so make sure to freshly copy your code or otherwise verify that it's actually running....

tannewt, 2023-05-08:
I don't see anything obvious that'd do it.

jepler, 2023-05-08:
I found that reverting the MIDI buffer size change fixed the problem here.  However, I'm also seeing the tip of main fail to boot on my metro m7...

jepler, 2023-05-08:
My bisect log so far, as I have to head out ...
```
git bisect start
# status: waiting for both good and bad commits
# bad: [b07de078f9dd640c28668cad3b5a5728a1d1024e] Merge pull request #7818 from CDarius/m5stack_stickc_fixes
git bisect bad b07de078f9dd640c28668cad3b5a5728a1d1024e
# status: waiting for good commit(s), bad commit known
# good: [cb6201d952c7be39024add8d91b79803ee191485] Merge pull request #7659 from dhalbert/gifio-displayio-doc
git bisect good cb6201d952c7be39024add8d91b79803ee191485
# bad: [461d833c1c9ead38542dbbfe665af5c3446035bd] Merge pull request #7763 from dhalbert/update-tinyusb-remove-usb_descriptor
git bisect bad 461d833c1c9ead38542dbbfe665af5c3446035bd
# good: [bca7d6e99933ef77af7cb29e5cb8c25d735051e1] Merge pull request #7717 from jepler/warn-too-new-uncrustify
git bisect good bca7d6e99933ef77af7cb29e5cb8c25d735051e1
# good: [22d7c0b8e904b5324155dbd668d41a8840b1d044] Merge pull request #7736 from bwshockley/fig-pi-update
git bisect good 22d7c0b8e904b5324155dbd668d41a8840b1d044
# good: [32a596df39ab8600cbc3d5f4ca4e0c65e5551033] Merge pull request #7757 from adafruit/8.0.x
git bisect good 32a596df39ab8600cbc3d5f4ca4e0c65e5551033
# good: [c78502d8a589a356e6eeb398762c62b910fc257f] Merge pull request #7746 from FoamyGuy/diskinfo_api
git bisect good c78502d8a589a356e6eeb398762c62b910fc257f
# bad: [7ee85408a1db9b79691692c74fa7ae10a0cf05b0] Merge pull request #7755 from isacben/block-device-arguments-typing
git bisect bad 7ee85408a1db9b79691692c74fa7ae10a0cf05b0
# bad: [67e0a49a1f29e87c4d7b35331dbb6c751b37ee49] Merge pull request #7724 from tannewt/get_perfbench_running
git bisect bad 67e0a49a1f29e87c4d7b35331dbb6c751b37ee49
# bad: [bdf592089ae461434a03ba0f185dfce3d09953fb] Fix .bin, .hex and .uf2 with new linker sections
git bisect bad bdf592089ae461434a03ba0f185dfce3d09953fb
# bad: [5bb8a7a7c68781e6edbd3aebaf7eeeecb310bd86] Improve iMX RT performance
git bisect bad 5bb8a7a7c68781e6edbd3aebaf7eeeecb310bd86
# first bad commit: [5bb8a7a7c68781e6edbd3aebaf7eeeecb310bd86] Improve iMX RT performance
# bad: [bd9aca2526b568eeacd8b8e02fe1a8d9baae3843] Merge pull request #7862 from jepler/synthio-envelope
git bisect bad bd9aca2526b568eeacd8b8e02fe1a8d9baae3843
# good: [8e00f7792823beda3a77bd48879825d888ff22d2] Merge pull request #7914 from adafruit/anecdata-patch-1
git bisect good 8e00f7792823beda3a77bd48879825d888ff22d2
# good: [8e00f7792823beda3a77bd48879825d888ff22d2] Merge pull request #7914 from adafruit/anecdata-patch-1
git bisect good 8e00f7792823beda3a77bd48879825d888ff22d2
# bad: [80e1418c01fadf434bc58bf3c8b0a0ba77ec9474] Merge pull request #7921 from tannewt/imx_evks
git bisect bad 80e1418c01fadf434bc58bf3c8b0a0ba77ec9474
# good: [c5d3de2a17cf23947cb07f9f61194196960edfe9] Merge pull request #7920 from bablokb/plasma2040w
git bisect good c5d3de2a17cf23947cb07f9f61194196960edfe9
# good: [1a4e1d114cc969aea251c1a2017de79628ac4ddb] Merge pull request #7916 from bablokb/badger_2040_w
git bisect good 1a4e1d114cc969aea251c1a2017de79628ac4ddb
```

PR #7953 (2023-05-12): Fix some regressions on feather M7
This closes #7944 but it's fishy -- why is such a large buffer required?  @hathach your feedback would be appreciated about this question.

It also includes a board-specific fix for #7952 but this may be needed on other boards. @tannewt do you know what's going on here? The USB_DM/USB_DP pins seem to just be dummy objects on 1011 but adding these to the list (as you did on the 1011 evk) does allow CP to start. If this is correct, the fix should be propagated to ALL boards, or incorporated at a better location.
```c
ports/mimxrt10xx/peripherals/mimxrt10xx/MIMX [... cut]
commit 5a5474b5 (2023-05-12): mimxrt: 101x: MIDI_RX_BUFSIZE must be 512

It's not clear why this is required, but otherwise it stops
midi_intest1 from ever receiving any MIDI events.

Closes: #7944
closed 2023-05-12 by tannewt