adafruit/circuitpython #7945: Defining only the x coordinate in a Vectorio Polygon with a float number crash CircuitPython

jposada202020, 2023-05-08:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.2 on 2023-04-26; Adafruit PyPortal Titano with samd51j20
```

### Code/REPL

```python
from vectorio import Polygon
import displayio
import board

display = board.DISPLAY
my_group = displayio.Group()
bitmap = displayio.Bitmap(300, 300, 5)
palette = displayio.Palette(1)
palette[0] = 0x125690
points = [(5, 5), (100, 20), (200.2)]
polygon = Polygon(pixel_shader=palette, points=points, x=0, y=0)
```

### Behavior

```python
Adafruit CircuitPython 8.1.0-beta.2 on 2023-04-26; Adafruit PyPortal Titano with samd51j20
>>> 
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Fault detected by hardware.
Please file an issue with your program at https://github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Adafruit CircuitPython 8.1.0-beta.2 on 2023-04-26; Adafruit PyPortal Titano with samd51j20
>>> 

```

### Description

This only happens if the number is a float.

If x value is an int it will raise `ValueError: point length must be 2`
if x, y are both floats it will raise `TypeError: x must be of type int, not float`

FoamyGuy, 2023-05-08:
I took a look into this and I think the ints vs. floats is not the root cause.

In the specific code you posted, this line:
```
points = [(5, 5), (100, 20), (200.2)]
```
The last item in the list gets evaluated as a number, not a tuple with a number inside of it. Since there is no comma, the parenthesis are being evaluated for order of operations rather than meaning tuple.

If you use the code like this (note the extra comma after the number):
```
points = [(5, 5), (100, 20), (200.2,)]
```
then it will successfully raise the exception about point length:
```
Traceback (most recent call last):
  File "code.py", line 14, in <module>
ValueError: point length must be 2
```
And it does the same whether you use an int or a float. 

I think the real root cause is at this line: https://github.com/adafruit/circuitpython/blob/7cd65ffb401a7ceb84a94f6d91b56b93569aa8d4/shared-module/vectorio/Polygon.c#LL39C1-L39C62 if something other than a tuple is passed in then it has the hard crash. With your code above it was a number instead of tuple and tripped this hard crash I think.

jposada202020, 2023-05-08:
Yes. Probably copy the last script, when i was trying to figure out the tuple and int raise messages😊

PR #7947 (2023-05-09): validate point type is tuple in vectorio Polygon
Attempt to resolve #7945 

Added validation inside of Polygon creation to ensure the values inside the points list are tuples before attempting to get values out of them.
closed 2023-05-09 by tannewt