adafruit/circuitpython #7956: analogbufio needs reset method

latkinso42, 2023-05-09:
### CircuitPython version

```python
analogbufio object returns bad data for multiple use of the method "readinto". This api needs a reset method. Problem is not supplied buffer. Issue is most likely that the assigned ADC and or DMA needs to be reset. We would like to be able to use the one instance of analogbufio.BufferedIn object multiple times. Currently its good for a one shot. A reset method, or automatic reset after "readinto" would not clear the supplied buffer, but it would reset the adc or re-init the adc and take care of pointing the dma to the beginning of the buffer. There may be more to the cleanup and reset than mentioned here.
```

### Code/REPL

```python
import analogbufio
import board 
mybuffer = array.array("H", 0x0000 for i in range(5000))
ab = analogbufio.BufferedIn(board.GP26, 480000)
ab.readinto(mybuffer) # good data
for i in range(5000):
   mybuffer[i] = 0
ab.readinto(mybuffer) # bad data
```

### Behavior

the second call produces bad data in mybuffer.

tannewt, 2023-05-10:
I agree that multiple calls should work. I don't think there needs to be an explicit reset.

What board and version of CircuitPython are you running on?

latkinso42, 2023-05-10:
Its a custom board using the RP2040, setup pretty much like the Pico with
4MB Flash and a jumper for the reset button.
Our product had to pause at CPy 8.0.4 (which is where we saw the issue). We
will upgrade in a few weeks.
-Lee