adafruit/circuitpython #7966: Crash when using @micropython.native decorator

rdlaner, 2023-05-12:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.2-2-g057c152d6-dirty on 2023-05-10; BPI-Leaf-S3 with ESP32S3
```

### Code/REPL

```python
>>> @micropython.native
... def test_fxn():
...     print("This is a test fxn")
...
...
...
>>> test_fxn()
```

### Behavior

Device crashes immediately after above code w/o any output. Upon reconnection I see the following message:

```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Fault detected by hardware.
Please file an issue with your program at https://github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

I've tried using the @micropython.native decorator in a variety of different ways from frozen code to using the repl. Each time, it crashes my device. In the `mpconfigboard.mk` file I have added the following definition which I thought should have allowed for this decorator to work:

`CIRCUITPY_ENABLE_MPY_NATIVE = 1`

Does CircuitPython not support the native and viper emitters for the esp32-s3?

### Additional information

I'm also building the latest 8.1.0 beta, but I haven't yet tried an older version since I need espnow for my project.

dhalbert, 2023-05-12:
We have not re-tried this in a long time, so it is not so surprising it may be broken -- sorry.

rdlaner, 2023-05-12:
Thanks @dhalbert. I'm happy to try and debug this further, I just haven't debugged Circuitpython much yet. Is there a way to get a crash log or backtrack or something else for when this happens? Just looking for a place to start investigating.

DavePutz, 2023-05-12:
Just as a point of info: this sample code worked on a ESP32-S3-DevKitC-1-N8 running 8.1 beta 2
```
Adafruit CircuitPython 8.1.0-beta.2-27-g0a3faf8c9 on 2023-05-12; ESP32-S3-DevKitC-1-N8 with ESP32S3
>>> @micropython.native
... def test_fxn():
...     print("This is a test fxn")
...
...
...
>>> test_fxn()
This is a test fxn
```

rdlaner, 2023-05-13:
Thanks @DavePutz ! So, after doing some git bisecting I ultimately found that the issue I'm having, ironically, corresponds to setting this definition in my boards mpconfigboard.mk file:
```
CIRCUITPY_ENABLE_MPY_NATIVE = 1
```

If I comment out that line I'm now able to use the @micropython.native decorator. I'm a little concerned that without that definition, though, the native emitter decorator isn't actually doing anything. Doing a quick grep I see that definition is disabled by default and that it does appear to be used for enabling native code:

```
> grep -Irin --exclude-dir="*build*" "CIRCUITPY_ENABLE_MPY_NATIVE" ./
.//py/circuitpy_mpconfig.h:70:#define MICROPY_EMIT_INLINE_THUMB        (CIRCUITPY_ENABLE_MPY_NATIVE)
.//py/circuitpy_mpconfig.h:71:#define MICROPY_EMIT_THUMB               (CIRCUITPY_ENABLE_MPY_NATIVE)
.//py/circuitpy_mpconfig.mk:220:CIRCUITPY_ENABLE_MPY_NATIVE ?= 0
.//py/circuitpy_mpconfig.mk:221:CFLAGS += -DCIRCUITPY_ENABLE_MPY_NATIVE=$(CIRCUITPY_ENABLE_MPY_NATIVE)
```

Based on this it does appear that the @micropython.native (and probably viper) decorators are broken. Unless there is another way to enable them that I can't tell.

rdlaner, 2023-05-15:
Quick update, I took a look at this PR and from what I can tell the only ISA supported by the native emitter is thumb:
https://github.com/adafruit/circuitpython/pull/2271/files

Naively, I tried a quick and dirty test where I force support for xtensawin by making this change to `circuitpy_mpconfig.h`:
```
//#define MICROPY_EMIT_INLINE_THUMB      (CIRCUITPY_ENABLE_MPY_NATIVE)
#define MICROPY_EMIT_INLINE_THUMB        (0)
//#define MICROPY_EMIT_THUMB             (CIRCUITPY_ENABLE_MPY_NATIVE)
#define MICROPY_EMIT_THUMB               (0)
#define MICROPY_EMIT_XTENSAWIN           (CIRCUITPY_ENABLE_MPY_NATIVE)
```

Unfortunately, that didn't seem to work, so I'm guessing there's more involved in supporting native code for xtensa.