adafruit/circuitpython #7971: Small synthio.Note issues

todbot, 2023-05-12:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.2-16-gf8cbdf88e on 2023-05-09; Adafruit QT Py RP2040 with rp2040
```

### Code/REPL

```python
import time, random
import board
import audiobusio, audiomixer, synthio

# qtpy rp2040 SPI pins be sure SCK pin is tied to Gnd
lck_pin, bck_pin, dat_pin  = board.MISO, board.MOSI, board.SCK

SAMPLE_RATE = 28000  # clicks @ 36kHz & 48kHz on rp2040

synth = synthio.Synthesizer(sample_rate=SAMPLE_RATE)
audio = audiobusio.I2SOut(bit_clock=bck_pin, word_select=lck_pin, data=dat_pin)
mixer = audiomixer.Mixer(voice_count=1, sample_rate=SAMPLE_RATE, channel_count=1,
                         bits_per_sample=16, samples_signed=True, buffer_size=2048 ) # buffer_size=4096 )
audio.play(mixer)
mixer.voice[0].play(synth)

# make and press note w/ vibrato
note = synthio.Note( frequency=110, vibrato_depth=0.5, vibrato_rate=5 )  # 110 Hz = A2
synth.release_all_then_press( (note,) )

vibrato_rate = note.vibrato_rate
print("vibrato_rate:", vibrato_rate)
time.sleep(3)  # let intitial vibrato sound
print("vibrato_rate:", vibrato_rate * 2)
note.vibrato_rate = vibrato_rate * 2

while True:
    time.sleep(0.1)
```

### Behavior

Vibrato starts out fine, but when `note.vibrato_rate` is set after note is sounding, vibrato goes away entirely.

jepler, 2023-05-12:
This is one of the issues that #7969 (earlier today) was intended to fix. can you test again?

todbot, 2023-05-12:
Oh yes, testing now!  Seems like `synthio.Note.vibrato_depth` and `synthio.Note.vibrato_rate` are gone?

todbot, 2023-05-12:
Oh wait I see. new API. nevermind, back to testing

todbot, 2023-05-12:
Yep, it's all good. With code to new API (see below), things work as intended.  Thanks!  And I get to close a bug! (a useless bug but still)

```py
# ... synth init as before ...

# make and press note w/ vibrato
note = synthio.Note( frequency=110, bend_mode=synthio.BendMode.VIBRATO )  # 110 Hz = A2 
synth.release_all_then_press( (note,) )

vibrato_rate = 5
vibrato_depth = 0.5
print("vibrato_rate & depth:", vibrato_rate, vibrato_depth)
note.bend_rate = vibrato_rate
note.bend_depth = vibrato_depth
time.sleep(3)  # let intitial vibrato sound
print("vibrato_rate:", vibrato_rate * 2)
note.bend_rate = vibrato_rate * 2

while True:
    time.sleep(0.1)
```

closed 2023-05-12 by todbot

jepler, 2023-05-12:
Thanks for the quick re-test!