adafruit/circuitpython #7974: Crash creating second IS31FL3741_FrameBuffer

akoebbe, 2023-05-13:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.2-19-g93b0e5b74-dirty on 2023-05-12; Adafruit QT Py ESP32-S3 no psram with ESP32S3
Board ID:adafruit_qtpy_esp32s3_nopsram
UID:4F21AF44318F
```

### Code/REPL

```python
import busio
import board
import time
import displayio
import is31fl3741
from adafruit_is31fl3741.adafruit_rgbmatrixqt import Adafruit_RGBMatrixQT
from adafruit_display_shapes.circle import Circle
from framebufferio import FramebufferDisplay

i2c = busio.I2C(sda=board.SDA1, scl=board.SCL1, frequency=1000000)

time.sleep(5)

WIDTH = 13
HEIGHT = 9
LEDS_MAP = tuple(
    (
        address
        for x in range(WIDTH)
        for y in range(HEIGHT)
        for address in Adafruit_RGBMatrixQT.pixel_addrs(x=x, y=y)
    )
)

displayio.release_displays()

is31_left = is31fl3741.IS31FL3741(i2c, addr=0x30)
is31_fb_left = is31fl3741.IS31FL3741_FrameBuffer(is31=is31_left, width=13, height=9, mapping=LEDS_MAP)
is31_fb_left.brightness = 0.1
display_left = FramebufferDisplay(is31_fb_left, auto_refresh=True)

is31_right = is31fl3741.IS31FL3741(i2c, addr=0x31)
is31_fb_right = is31fl3741.IS31FL3741_FrameBuffer(is31=is31_right, width=13, height=9, mapping=LEDS_MAP) # <--- breaks here
display_right = FramebufferDisplay(is31_fb_right, auto_refresh=True)

shape = Circle(6, 4, 3, fill=0x00FF88)
group = displayio.Group()
group.append(shape)

display_left.show(group)

while True:
    print(".", end="")
    time.sleep(1)
```

### Behavior

```
press Ctrl-C to enter the REPL

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Fault detected by hardware.
Please file an issue with your program at https://github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 8.1.0-beta.2-19-g93b0e5b74-dirty on 2023-05-12; Adafruit QT Py ESP32-S3 no psram with ESP32S3
>>> 
```

### Description

Microcontroller: Adafruit QT Py ESP32-S3 no psram

I've customized Circuitpython to enable `CIRCUITPY_IS31FL3741` and increased the `CIRCUITPY_DISPLAY_LIMIT` to 2. I have two IS31FL3741 13x9 RGB matrices (difference addresses) chained on I2C. I'm trying to get both to work with displayio to show separate displays (not mirrored). I'm able to get one or the other to work alone (with both connected), but as soon as I try to create the second `is31fl3741.IS31FL3741_FrameBuffer` I get a crash.

CP modifications
```diff
diff --git a/ports/espressif/boards/adafruit_qtpy_esp32s3_nopsram/mpconfigboard.h b/ports/espressif/boards/adafruit_qtpy_esp32s3_nopsram/mpconfigboard.h
index f6e51ea58..7849aabbc 100644
--- a/ports/espressif/boards/adafruit_qtpy_esp32s3_nopsram/mpconfigboard.h
+++ b/ports/espressif/boards/adafruit_qtpy_esp32s3_nopsram/mpconfigboard.h
@@ -43,3 +43,5 @@
 #define CIRCUITPY_BOARD_UART_PIN    {{.tx = &pin_GPIO5, .rx = &pin_GPIO16}}
 
 #define DOUBLE_TAP_PIN (&pin_GPIO10)
+
+#define CIRCUITPY_DISPLAY_LIMIT (2)
\ No newline at end of file
```
```diff
diff --git a/ports/espressif/boards/adafruit_qtpy_esp32s3_nopsram/mpconfigboard.mk b/ports/espressif/boards/adafruit_qtpy_esp32s3_nopsram/mpconfigboard.mk
index e695bae63..5197cf7b4 100644
--- a/ports/espressif/boards/adafruit_qtpy_esp32s3_nopsram/mpconfigboard.mk
+++ b/ports/espressif/boards/adafruit_qtpy_esp32s3_nopsram/mpconfigboard.mk
@@ -14,3 +14,4 @@ FROZEN_MPY_DIRS += $(TOP)/frozen/Adafruit_CircuitPython_Requests
 FROZEN_MPY_DIRS += $(TOP)/frozen/Adafruit_CircuitPython_NeoPixel
 FROZEN_MPY_DIRS += $(TOP)/frozen/Adafruit_CircuitPython_Register
 CIRCUITPY_ESPCAMERA = 0
+CIRCUITPY_IS31FL3741 = 1
\ No newline at end of file
```

RetiredWizard, 2023-05-13:
Probably not directly applicable but incase you're interested, foamyguy poked at a very similar issue for a couple hours on his [4/7/23 youtube deep dive](https://www.youtube.com/watch?v=TPXBw9MGBbs). I think he was using SPI displays though.

akoebbe, 2023-05-14:
Thanks for the lead @RetiredWizard!

@FoamyGuy I just watched your live stream. It sounds kinda similar to what I'm experiencing, but I'm not even able to get far enough to even try to get the second displayio display running. This one fails when initializing the IS31FL3741 object. This is I2C on one bus, though I'll try the ESP's second i2c bus for the second matrix and see if that makes a difference.

Did you happen to get any farther with your issue on the monster m4sk?

akoebbe, 2023-05-14:
I tried using the other i2c bus for the second screen. No luck, same behavior. Each would work on their own, but not at the same time.

One interesting thing though, when I tried to break out of the code in to repl, it gave me an empty stack trace and restarted in safe mode. I'm going to try removing my build configs (2 display limit and native IS31) and see if that continues to happen.

FoamyGuy, 2023-05-14:
@akoebbe I got some issues relating to multi-display resolved, but I think there are still some remaining. IIRC The Monster M4sk was working to initialize both displays when I last messed with it. But I did still have some hard faults with other combinations of devices and displays. I don't believe I ever tried specifically two I2C displays, and none of my testing was with FrameBuffer display / RGB Matrix.

furbrain, 2023-05-15:
I believe this may be an implication of bug #2204
`gc_never_free` crashes on the fourth time it is called and the IS31FL3741 code calls `gc_never_free` twice for each display

akoebbe, 2023-05-16:
@furbrain I've confirmed that your fix resolves my issue. I'm able to get both displays running on the same bus with different images. Not sure if it's related, but when ever auto-reload kicks in, it crashes on boot. Hitting the reset button does get it back going again, but I just wanted to mention that in case it might be a side effect of the fix.

dhalbert, 2023-05-16:
Fixed by #7983.

dhalbert, 2023-05-16:
Probably fixed by #7983. Needs re-test.

akoebbe, 2023-05-17:
@dhalbert Retested and works fine with #7983. I have new things to troubleshoot, but that will most likely be a different issue.

closed 2023-05-17 by akoebbe