adafruit/circuitpython #7978: Nano Connect RP2040 - Dualbank support

dextermarkley, 2023-05-14:
The Nano Connect RP2040 has 16mb of flash and may be able to support [dualbank](https://docs.circuitpython.org/en/latest/shared-bindings/dualbank/index.html). 

Does anyone know of any blockers for implementing this?

dextermarkley, 2023-05-15:
The dependency rabbit hole goes a deeper than I hoped for each port.  

[esp_ota_ops.c](https://github.com/adafruit/esp-idf/blob/630c2724fc8c69eeaaa1bb025de52b99c5cb11aa/components/app_update/esp_ota_ops.c) looks port specific and a significant chunk of code that is beyond my C expertise.

layogtima, 2023-09-30:
@tannewt Does the RO2040 Nano support Circuit Python over the web workflow or is that like an ESP32 thing? A friend just got one and we've been struggling to get REPL working over the web workflow.

anecdata, 2023-09-30:
web workflow is supported on ports with native wifi... that's ESP32 variants (as the main MCU), and Pico W

tannewt, 2023-10-02:
> @tannewt Does the RO2040 Nano support Circuit Python over the web workflow or is that like an ESP32 thing? A friend just got one and we've been struggling to get REPL working over the web workflow.

RP2040 Nano is not supported because it is an RP2040 connected to an ESP32. (We call this an airlift for adafruit products.) We are moving to a single core solution with ESP. So, I don't expect we (Adafruit-funded folks) will ever add it.

layogtima, 2023-10-02:
@anecdata @tannewt Appreciate the context and information!

Very new to hardware, and community help has been very helpful. 

Have a great week ahead!