adafruit/circuitpython #7979: Serial monitor does not work on YD-ESP32-S3 N16R8 version 8.1.0 beta2 and 20230512-827eaeb

wuhao19920619, 2023-05-14:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.2-31-g827eaeb1f on 2023-05-12; VCC-GND YD-ESP32-S3 (N16R8) with ESP32S3
Board ID:yd_esp32_s3_n16r8
UID:4F21AF1E1F4B
```

### Code/REPL

```python
print("hello world!")
```

### Behavior

My IDE is VScode with Circuitpython extension.

after `>CircuitCython: Open Serial Monitor` , a serial port selection window pops up with the following information: `COM20 Microsoft | 303A | 8166`. The information followed by `COM20` seems nonsensical. and when I click on it, nothing happens.

### Description

I have tested RP2040 and ESP32-S3-BOX V8.0.5, and both devices worked perfectly. Therefore, the issue is not with Visual Studio Code or the CircuitPython extension.

casainho, 2023-05-15:
I also use VScode and on Linux.

Are you sure your serial port exists on your system? can you open the serial port on any terminal software??

bill88t, 2023-05-15:
You should try connecting to the board with Putty for testing.

You can find the COM port of the board from the windows device manager (Super + X -> Device Manager).
Remember to set the baudrate to `115200`.

I am the one who made the board port, and it works on my sample.

wuhao19920619, 2023-05-15:
@bill88t @casainho I have tested with putty, rather than some unreadable code, all others work good. But serial monitor in circuitpython still doesn't work.
[image]

Also,  I did some test with the ESP32-S3-BOX V8.0.5. Once I close the Circuitpython cmd window in the vscode, the commands `>circuitpython: Select Serial Port` , `>circuitpython: Open Serial Monitor` and `>circuitpython: Close Serial Monitor` will no longer work. The command window will not pop up. So I am not sure if this issue comes from the circuitpython extension.

By the way, I use VScode on Windows 10 Home.

bill88t, 2023-05-15:
If the REPL is working (you can read and write to it) the issue does not lie within circuitpython.
Raw REPL (can be opened with Ctrl + A), that these programs use also works on my sample as expected:
```py
>>> 
raw REPL; CTRL-B to exit
>OKHello world!

>
```

Taking a look at the [circuitpython extension](https://marketplace.visualstudio.com/items?itemName=joedevivo.vscode-circuitpython), it warns:

```
NOTE FOR WINDOWS USERS: I have seen trouble with the serial console, displaying anything at all. If that happens, try launching VSCode as an administrator and see if it works. I have even gotten it to work as a non-administrator after this, so perhaps running it as an admin stole the serial port from whatever was using it, and then whatever it was didn't grab it again.
```

In any case, this really isn't a circuitpython issue.
If you can read and type at the REPL and the CIRCUITPY drive, the board and it's definition are working as intended.

You should go ahead and report it in https://github.com/joedevivo/vscode-circuitpython if this is indeed your plugin.

wuhao19920619, 2023-05-15:
Yes, I agree there must be some issues with vscode-circuitpython. But I am just curious why that serial monitor only works with verson 8.0.5.

I tried putty again but this time it stucks, displaying nothing at all and no responding to any keyboard input. I connected with correct COM port and BAUD rate 115200. There is some information on the top bar of the window.
[image]

ouening, 2023-09-01:
I have the same problem!
**Firmware**: `adafruit-circuitpython-yd_esp32_s3_n16r8-en_GB-8.2.4.bin`
**Dev board:** YD-ESP32-S3 N16R8, 2022-v1.3
After flashing the firmware and connected to it using putty, REPL did not work:
[image]

bill88t, 2023-09-02:
@ouening Are you connected to the port labeled as "USB" on the board? The "COM" isn't gonna show the REPL.

ouening, 2023-09-06:
> @ouening Are you connected to the port labeled as "USB" on the board? The "COM" isn't gonna show the REPL.

Yes, when I connected to the 'USB' port, it works! Thank you!

dhalbert, 2023-09-06:
@StreakingJerry @casinho are you also connecting to the port labeled "USB" on the board, as discussed above?

closed 2023-09-06 by dhalbert

casainho, 2023-09-07:
Everything working well for me, you know, very easy to check on Linux that the USB Serial Port is working as also the USB Mass Storage device:

[image]

[image]