adafruit/circuitpython #7980: Add Luatos Core ESP32S3, ESP32C3 Board

wuhao19920619, 2023-05-14:
Hi,

Those 2 Luatos boards are quite cheap and simple. Could you please add support to them? Thank you.

The schematics for luatos ESP32-S3:
[https://wiki.luatos.com/chips/esp32s3/hardware.html](url)
Note: the MCU is ESP32-S3R8, the flash could be 25lq128ewig or 25lq128cvig

There are two version of ESP32-C3:
[https://cdn.openluat-luatcommunity.openluat.com/attachment/CORE-ESP32-C3%E8%AE%BE%E8%AE%A1%E5%8E%9F%E7%90%86%E5%9B%BE_V1.2.pdf](url)
[https://cdn.openluat-luatcommunity.openluat.com/attachment/20220609213416069_CORE-ESP32-A12.pdf](url)

dhalbert, 2023-05-14:
Hi - Adafruit staff does not perform the work of adding boards that we don't sell. We welcome board additions by the community: https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython

wuhao19920619, 2023-05-15:
Thank you for the information. I have notified the manufacturer, and I believe they will add the support very soon.

closed 2023-05-15 by wuhao19920619