adafruit/circuitpython #7991: Arduino Nano 33 BLE Sense board def?

applecuckoo, 2023-05-17:
This could be based on the existing Arduino Nano 33 BLE definition, just with definitions for `board.MICROPHONE_CLOCK` and `board.MICROPHONE_DATA` which are connected to the PDM mic on the board. Probably something similar to the nRF52 Sense Feather since they are both similar ideas, and maybe a modified version of the guide could go with it. I've actually discussed this with an old account at https://github.com/adafruit/circuitpython-org/issues/410.

dhalbert, 2023-05-17:
If you'd like to contribute a board definition, that would be great. In the other issue you mentioned you don't own the board. Do you have one now?

applecuckoo, 2023-05-17:
@dhalbert No, I don't own this board. I could probably start from the existing definition, but I should probably get my current PR (#7989) merged before I start work on this.

applecuckoo, 2023-05-18:
I've specifically changed the scope of this issue to focus on the newer Rev 2 version of the board since the existing definition incorporates the sensor pins in pins.c, as seen here:
https://github.com/adafruit/circuitpython/blob/f2bfced4079389c71cbdcf7d69e16d41f0e3dc98/ports/nrf/boards/arduino_nano_33_ble/pins.c#L51-L55
There are a few differences between the two revisions pin-wise, here is a list of them that I've picked up:
- The LPS22's interrupt pin is now connected to the board at P0.12
- The LSM9DS1 has been substituted for two sensors combining to form a 9-DoF IMU.
  - One of these sensors, the Bosch BMI270, has two interrupt pins connected at P0.11 and P0.20 respectively. This isn't very useful at the moment since a CircuitPython library doesn't exist for the BMI270, but it could be handy if one ever gets developed.
- There are also two pins that you can turn off to disable sensors:
  - p0.22 for the IMU, LPS22 and HS3003 humidity sensor
  - And p0.17 for the mic.

I hope this helps for anyone making a board definition for the Rev 2.

applecuckoo, 2024-06-20:
Closed by #9347.

closed 2024-06-20 by applecuckoo