adafruit/circuitpython #7993: make -C mpy-cross fails, no "huffman"

todbot, 2023-05-17:
### CircuitPython version

```python
n/a
```

### Code/REPL

```python
make -C mpy-cross
```

### Behavior

% make -C mpy-cross
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/moduledefs.h
QSTR updated
Traceback (most recent call last):
  File "/Users/tod/projects/adafruit/circuitpython-adafruit/mpy-cross/../py/maketranslationdata.py", line 30, in <module>
    import huffman
ModuleNotFoundError: No module named 'huffman'
make: *** [build/genhdr/compression.generated.h] Error 1

### Description

Looks like doing "pip3 install huffman" fixes this.

So add this to `requirements-dev.txt" ?

tannewt, 2023-05-18:
We use an in-tree version of `huffman` but I don't remember why. This bit of code is meant to add it to the path: https://github.com/adafruit/circuitpython/blob/main/py/maketranslationdata.py#L26-L30

Try building from the mpy-cross directory.

todbot, 2023-05-18:
oh wait no `cd mpy-cross && make` does not work.

todbot, 2023-05-18:
for some reason `make fetch-submodules` does not pull in "tools/huffman". I had to explicitly get it with `cd tools/huffman; git checkout`

tannewt, 2023-05-19:
> for some reason `make fetch-submodules` does not pull in "tools/huffman". I had to explicitly get it with `cd tools/huffman; git checkout`

Ah! That would do it.

dhalbert, 2023-05-22:
> for some reason `make fetch-submodules` does not pull in "tools/huffman". I had to explicitly get it with `cd tools/huffman; git checkout`

I did a fresh clone of my fork, and did `make fetch-submodules`. `huffman` got cloned and populated as I would expect. I am using git 2.40.1 on Ubuntu 22.04. I use the git PPA to get a newer version than 22.04 would normally provide.

Maybe there is something wonky about your clone. You could try `git remove-submodules` and then `git fetch-submodules` again, or just try a fresh clone.

todbot, 2023-05-23:
This is weird. I have also been building on an Ubuntu 22 x86_64 box and I swear I saw it there too. 
But I just did a fresh checkout on Ubuntu (git 2.34.1) and macos (homebrew git 2.39.1) and I am not experiencing this issue now. I do know that someone else on a Mac I was helping build a board definition had the same problem and they perhaps did not have homebrew git.  And at one point during this I was having PATH issues wrt homebrew. 
Closing this issue because I think the problem was me.

closed 2023-05-23 by todbot

m-mcgowan, 2025-02-20:
I saw this problem recently. I was running make 3.81, while the [installation guide for OSX](https://learn.adafruit.com/building-circuitpython/macos) recommends 4.3 or newer. 

`brew install make` fixed it up for me.

ericzundel, 2025-06-27:
Fleshing out this issue for posterity.

I replicated this problem today when building on WSL2/Ubuntu.  For whatever reason, I'm getting timeout errors from github trying to run `make fetch-submodules-port`. The errors when running the build look similar to this:

```
fatal: unable to access 'https://github.com/georgerobotics/cyw43-driver/': SSL connection timeout
fatal: could not fetch 348725336b51b158b7eaca8bd53113555cfa6093 from promisor remote
fatal: Unable to checkout 'c1075d4bc440422cf2b2fd12c64a1f53f77660ee' in submodule path 'ports/raspberrypi/lib/cyw43-driver'
::endgroup::
Duration: 302.97863835300086
Traceback (most recent call last):
  File "/home/zundel/replicate-7933/circuitpython/ports/raspberrypi/../../tools/ci_fetch_deps.py", line 263, in <module>
    main(target)
  File "/home/zundel/replicate-7933/circuitpython/ports/raspberrypi/../../tools/ci_fetch_deps.py", line 245, in main
    fetch(submodules)
  File "/home/zundel/replicate-7933/circuitpython/ports/raspberrypi/../../tools/ci_fetch_deps.py", line 131, in fetch
    run(
  File "/home/zundel/replicate-7933/circuitpython/ports/raspberrypi/../../tools/ci_fetch_deps.py", line 113, in run
    subprocess.run(shlex.split(command), stderr=subprocess.STDOUT, check=True, cwd=cwd)
  File "/home/zundel/miniconda3/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['git', 'submodule', 'update', '--init', '--filter=blob:none', 'data', 'extmod', 'lib', 'tools', 'frozen', 'ports/raspberrypi', 'extmod/ulab/', 'lib/adafruit_floppy/', 'lib/mbedtls/', 'lib/mp3/', 'lib/certificates/', 'lib/protomatter/', 'lib/quirc/', 'lib/tinyusb/', 'lib/tlsf', 'data/nvm.toml/']' returned non-zero exit status 128.
make: *** [../../py/circuitpy_defns.mk:973: fetch-port-submodules] Error 1
```

When that happens, re-running `make fetch-submodules-*` is unlikely to fix the problem as `git submodule update` doesn't repair half-cloned submodules as is well documented on Stack Overflow and other places.  To fix this particular issue I tried some of the Stack Overflow suggestions to add flags to git submodule update, but they didn't help.  One thing that did work was 

```
cd tools/huffman
git reset --hard
git pull
```

which fixed  that submodule and allowed me to build mpy-cross.  I'm getting timeouts repeatedly and fear there are others that did not get pulled correctly.  I'm not sure what a good general solution is other than removing all submodules as suggested by dhalbert above and trying again at another time.