adafruit/circuitpython #7994: TouchAlarm on Qt Py ESP32-S2 not working in SLEEP mode

mikeysklar, 2023-05-18:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-rc.0 on 2023-05-17; Adafruit QT Py ESP32S2 with ESP32S2
```

### Code/REPL

```python
import alarm
import board

print(alarm.wake_alarm)

touch_alarm = alarm.touch.TouchAlarm(board.A2)

alarm.exit_and_deep_sleep_until_alarms(touch_alarm)
```

### Behavior

The code runs, but does not wakeup upon pin touch.

### Description

[Forum thread](url=https://forums.adafruit.com/viewtopic.php?t=201048) based on example [Adafruit example code](url=https://learn.adafruit.com/deep-sleep-with-circuitpython/alarms-and-sleep#timealarm-deep-sleep-3078700).

rxhfcy, 2023-05-18:
Consider adding the esp32-s2 label?

dhalbert, 2023-05-18:
This appears to have stopped working between 7.1.0 and 7.2.0. I'll bisect.

dhalbert, 2023-05-18:
Apparently due to 13db65566d590fea872121d51eeacfc5d2d612a2:
> Pin reset is now changed to the IDF default which pulls the pin up rather than CircuitPython's old behavior of floating the pin.

So we need to special-case the touch pins

dhalbert, 2023-05-19:
2bc8d73776434bb2f15113014299a9fffa547109, which is past 13db655, _also_ causes `TouchAlarm` not to work. If I prevent the touch awaking pin from being reset, then 13db655 and forward works, up to but not including 2bc8d73776434bb2f15113014299a9fffa547109 does not. 2bc8d73776434bb2f15113014299a9fffa547109 is part of #6772.

So still working on this.

PR #8003 (2023-05-22): Esp touch alarm fix
- Fixes #7994.

Two problems:
- #5892 made pin reset set a pull-up, which is recommended by Espressif for minimal power draw (and prevents glitches on random pins). Needed to not set the pull-up to make touch waking work. Added a new capability in `microcontroller/Pin.c` to not reset just once. I was originally using never-reset to do this, but I can't always clear never-reset at the right time, so sometimes it sticks around and causes the pin to appear to be in use.
- #6772 fixed touch on plain ESP32 and switched from `TOUCH_FSM_MODE_TIMER` to `TOUCH_FSM_MODE_SW`. `SW` doe [... cut]
closed 2023-05-22 by tannewt