adafruit/circuitpython #7995: Please add ulab.numpy.absolute function

jamesra, 2023-05-18:
The [abs](https://numpy.org/doc/stable/reference/generated/numpy.absolute.html) function from numpy would be nice to add to ulab.numpy.  There is a lot of numpy functionality missing for good reason but this function feels particularly fundamental and it would be convenient to have it built-in.  Thanks.

v923z, 2023-05-19:
@jamesra The absolute value unary operator was one of the first operators implemented: 

https://github.com/v923z/micropython-ulab/blob/3e996d9bd93005edbbb9763c4a0a525fe52dec3d/code/ndarray.c#L1968-L2000
 Or do you have something else in mind?

v923z, 2023-05-19:
OK, but then is it worth the flash space? If there is a good reason, we can easily add the function.

v923z, 2023-05-19:
So, if this is the function in question https://numpy.org/doc/stable/reference/generated/numpy.fabs.html, then the difference to `abs` is that it can support keyword arguments, which are not supported in `ulab`, anyway. The only useful argument there is `out`, I believe, which we could add. But beyond that, I don't see any advantages compared to `abs`.

jamesra, 2023-05-19:
I don't have hardware handy to double-check, but this may be a [documentation](https://docs.circuitpython.org/en/latest/shared-bindings/ulab/numpy/index.html) issue instead.  abs is only listed on the page for complex arrays.  I tried np.abs and np.absolute, but I know I did not try calling it directly off my array.  Since I don't have a debugger on CircuitPython I'm not able to use the normal method of dir(myarray) to discover what is available.  np.abs is probably still worth having, but not necessary if it is exposed on the array object itself.  (My apologies if abs is documented somewhere else and I just missed it.)

jamesra, 2023-05-19:
OK, at the risk of sounding dumb (which I already know, thanks) how is the unary abs operator used?  I did not find the abs function on the array in an interactive console:

jamesra, 2023-05-19:
OK, at the risk of sounding dumb, what is the syntax for the unary absolute value operator in numpy?  (I am using 8.1.0-beta 2 to obtain analogbufio on the ESP32-S3)

I launched an interactive Circuit Python session and there is no abs function on the array:
```
Adafruit CircuitPython 8.1.0-beta.2 on 2023-04-26; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
>>> import ulab.numpy as np
>>> a = np.array([1,-2,3,-4])
>>> dir(a)
['__class__', 'copy', 'sort', 'byteswap', 'dtype', 'flat', 'flatten', 'itemsize', 'reshape', 'shape', 'size', 'strides', 'tobytes', 'tolist', 'transpose']
>>> a.abs
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'ndarray' object has no attribute 'abs'
>>> a.absolute
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'ndarray' object has no attribute 'absolute'
```

jepler, 2023-05-19:
```py
MicroPython 8.1.0-beta.2-43-g4901bdceb7-dirty on 2023-05-18; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import ulab.numpy as numpy
>>> abs(numpy.array([-1, 2, -3]))
array([1.0, 2.0, 3.0], dtype=float64)
```
It corresponds to the abs built-in function. In another post v923z referred to it as a "unary operator", but that reflects some implementation detail of micropython, it's not actually a unary operator like `-` is in a Python statement like `x = -y`.

jamesra, 2023-05-19:
Oh!!! I'm trying to decide how I feel about that.  On one level it makes total sense to use the built-in language operator.  On another level it is some lateral thinking to step outside of the numpy library to do one operation... My suggested fix would be documentation.

v923z, 2023-05-19:
> My suggested fix would be documentation. Thank you for explaining the intended use.

This has actually been documented, unless you have something else in mind: https://micropython-ulab.readthedocs.io/en/latest/ulab-ndarray.html#unary-operators

dhalbert, 2023-11-07:
Closing because I think this has been resolved.

closed 2023-11-07 by dhalbert