adafruit/circuitpython #8008: Make safemode.py etc as part of basic installation - ask anecdata for Gist

tyeth, 2023-05-20:
@anecdata 's [stuff ](https://gist.github.com/anecdata/cfc4c585406d17985885bfe3a1cc9c73) is cool, I wish it was standard functionality, like came for free with circuitpython. (add a safemode.py that appends the reset reason to a file, the boot.py to catch errors and check reason for bootup, and a code.py with the 3lines printing boot reason plus something circuitpythony/hello-worldy)
It's linked from some general advice about programming for [long-term reliability on circuitpython here](https://gist.github.com/anecdata/cfc4c585406d17985885bfe3a1cc9c73).
[The safemode.py gist](https://gist.github.com/anecdata/fe35dc6a94069fc920edf61a64750b53) has 3 files at the top, plus a bit nearer the bottom, and would need a minor adjustment or two, but basically it's really unfriendly for everyone when something goes wrong and it's not recorded as to why or what went wrong, so we could all benefit massively from this one.
 There could even be a new adafruit IO example that submits new reset reasons to an error feed with device-name/mac-address included.

tannewt, 2023-05-22:
@dhalbert may be worth integrating into the safemode.py guide.

dhalbert, 2023-05-22:
@tyeth I'm assuming you read https://learn.adafruit.com/circuitpython-safe-mode. Without safemode.py, the reason is printed in the REPL. The original motivation of safemode.py was not to do a more thorough diagnosis, but to recover even in the presence of a safemode.

I can add more detailed examples to the guide. I'd rather not complicate the beginner's experience with a pre-programmed safemode.py.

tyeth, 2023-05-23:
@dhalbert 
> I'm assuming you read https://learn.adafruit.com/circuitpython-safe-mode. 
Sadly not :) I think I had in the past I believe, but re-reading it now there's a few useful things I picked up. I wonder if another example showing a more complex case or an external link to a repo/project/learn-guide with bundle with the "log all the things locally (exceptions and reboot reasons with time/traceback) and then to adafruit IO after reboots (or immediately if recoverable error? I'm just using microcontroller.reset for most errors). Alternatively another example that shows the logging of the reasons like anecdata's gist would be useful even if it's not much of a leap for users to construct their own.

Having digested a bit on the idea, I realise what I want is maybe a useful starter template, probably it's a separate repo and just suggested if you are experiencing reboots / errors and want them logged or at the end of the reading safemode.py guide.
 I imagine hello world followed by a divide by zero error, and the whole lot wrapped in something that saves the timestamp/stacktrace/exception (is sleepmemory/nvm enough for that) to rtc-mem or file if possible/desired (not sure how remounting would affect errors or serial but ideally just remount to write files and remount to read-only again). All protected by a button check(is boot only user button on every board?) in code.py/boot.py to activate usb storage instead of the main routine that offers no usb storage access (maybe unnecessary if remount has little side-effect for most use-cases).
 
 Also the safemode guide could be enhanced to suggest [the page for reading about the safemode button combination](https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting#circuitpython-7-dot-0-0-and-later-2978455) for your board. Maybe on the front page where it lists reasons in bullet points, the "pressed one or more buttons" phrase could be linked. I'd never managed to trigger safemode except from REPL before reading the guide just now, boot always was for CDC mode in my mind, and double reset for uf2 bootloader. Having now seen the flowchart I see there is a 1second wait for boot button :)

tyeth, 2023-06-11:
Not got around to creating a clean starter template yet, but this now has all the features I was looking for (except forwarding previously unforwarded errors to MQTT feed - TODO).
- Logs errors at boot / safemode, plus Reason + traceback + timestamp.
- Traps errors in code.py and saves to NVM before rebooting to save to flash filesystem in appropriate json file
- Configurable error/boot/safemode log lengths, with free disk space checking and file rollover (If not enough space to add entry, then cleanup current file which keeps last X entries if possible else removed and new entry added).

I appreciate one can disable usb storage and then the code.py file could write to storage at any time, but in the development cycle one needs to quickly iterate and having code.py and errors.json accessible to the computer is more valuable.

https://github.com/good-enough-technology/Good-Enough_Air-Quality-Device_CO2-SCD4x_PM-SEN5x_CircuitPython

DJDevon3, 2023-06-12:
I like the idea of an advanced example.  I could certainly find it useful especially when beta testing and running into random hard faults or exceptions. A quick way to log errors to help myself and/or developers could be handy.  It depends on the board though, some boards like an M0 just won't have much if any room for that kind of thing.  I like the general idea.