adafruit/circuitpython #8013: Hidden elements drastically slow down displayio refresh

ventrue3000, 2023-05-21:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-rc.0 on 2023-05-17; Teensy 4.0 with IMXRT1062DVJ6A
Adafruit CircuitPython 8.0.5 on 2023-03-31; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
# Import modules
import board
import busio
import displayio
import time
from adafruit_display_shapes.rect import Rect
from adafruit_display_shapes.line import Line	
from adafruit_st7789 import ST7789

# Pin definitions for Teensy 4.0
PinSPISCK=board.D13
PinSPICOPI=board.D11
PinSPICIPO=board.D12
PinDisplayCS=board.D10
PinDisplayRST=board.D15
PinDisplayDC=board.D16
PinDisplayBacklight=board.D7

# Set up SPI
displayio.release_displays()	# Displays live on after soft resets. Not releasing it before initialising SPI throws an error after soft resetting
SPI=busio.SPI(PinSPISCK, PinSPICOPI, PinSPICIPO)

# Set up display - 240x280 ST7789
DisplayBus=displayio.FourWire(spi_bus=SPI, command=PinDisplayDC, chip_select=PinDisplayCS, reset=PinDisplayRST)
Display=ST7789(DisplayBus, width=240, height=280, rotation=0, rowstart=20, auto_refresh=False, backlight_pin=PinDisplayBacklight, brightness=1)
DisplayContent=displayio.Group(x=0, y=0)

# Create lines to indicate speed
DisplayDemoRects=[]
for i in range(70):
	DisplayDemoRects.append(Rect(x=0, y=i*4, width=240, height=4, fill=0xFFFFFF))
	DisplayContent.append(DisplayDemoRects[i])

# Create additional elements to slow everything down (or not, ideally)
DisplayDemoBallast=[]
for i in range(150):
	DisplayDemoBallast.append(Line(x0=0, y0=0, x1=100, y1=100, color=0xFFFFFF))
	DisplayDemoBallast[i].hidden=True	#FIXME Commenting out this line solves the lag issue
	DisplayContent.append(DisplayDemoBallast[i])

# Show everything
Display.show(DisplayContent)
Display.refresh()

# Main loop - moves across the screen and re-colours one indication line after the other
DemoColour=0x000000
while True:
	print(time.monotonic())
	for i in range(70):
		DisplayDemoRects[i].fill=DemoColour
		
		# Refreshing manually improves things substantially
		Display.refresh()

	# Colour shift for the next pass...
	DemoColour+=1600000

	# ... but don't exceed what's possible
	if DemoColour>0xFFFFFF:
		DemoColour-=0xFFFFFF
```

### Behavior

Changes on the display happen incredibly slowly. One pass of the `while` loop at the end takes four minutes!

Issue gets more severe when more things are hidden.

No errors.

### Additional information

Commenting out the line that hides elements (marked with `#FIXME`) solves the issue. One pass of the `while` loop at the end then takes 0.75 seconds.

Example above is for a Teensy 4.0 with a 240x280 ST7789 display, connected with SPI.
With the necessary changes to the code, this can also be reproduced on:
- a 240x320 ST7789 display.
- a 240x320 ILI9341 display.
- on a Pico W with a 240x280 ST7789 display.

Maybe interesting: The LED connected to the SCK pin on the Teensy is always on when the issue occurs.

tannewt, 2023-05-22:
Are you interested in hunting this down? I'm happy to help guide you.

ventrue3000, 2023-05-22:
I'd happily help out where I can, but based on what I've seen of CP's code, I'd say that even with help, solving this issue is quite a bit beyond my abilities :-/

But in the mean time, as a workaround, objects can simply be moved by a fixed amount off and back onto the screen. This does not change their order in the Z direction and preserves the original coordinates.

FoamyGuy, 2023-05-26:
I have successfully re-recreated this a learned a little bit about it, but still need to dig a bit further and do not have an actual solution to propose.

A few findings from my testing:

- Line object makes it go much slower than Rect objects. I tested by changing the 150 loop to make Rects instead of Lines and the display refreshes much faster in that case. (even if I make the Rects be 100x100 which is the size of the Bitmap for the Diagonla Lines)
- Line objects are not just inherently slow though, it also depends on the size / shape of the line. A fully horizontal or fully vertical line allows the display to refresh much faster than the diagonal line. (which makes sense because the diagonal line is least efficient since it has to make a Bitmap big enough to contain the full rectangle that encloses the diagonal line and many pixels will be transparent, whereas with horizontal or vertical, the Bitmap gets to be the exact size it needs to be in order to show the line without any transparent pixels)
- The place where we are losing time seems to be inside of this while loop: https://github.com/adafruit/circuitpython/blob/0aaf5a4a98c38d376ebc82acaf041e685a87327d/shared-module/displayio/Display.c#L333-L336 and strangely we get iterations within this loop for things that are `hidden`. So when the Lines are hidden we are still iterating through them once per each line in this loop calling _refresh_area on them. I'm thinking that may be a logic bug somewhere because I would think we could skip them in the refresh loop if they are hidden. I'm trying to chase a bit further into `_get_refresh_areas()` which is where they come from before it starts looping over them, but I'll have to dig further another time, I've got to stop for now.

closed 2023-06-21 by tannewt