adafruit/circuitpython #8019: ValueError: BUTTON_L in use

phamhuyvn, 2023-05-22:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-rc.0 on 2023-05-17; LILYGO T-DISPLAY with rp2040
Board ID:lilygo_t_display_rp2040
UID:DE62309247682B26

I can't use board.BUTTON_L , and if I check with code below. I see :

board.BUTTON_L board.BUTTON_R board.GP6 , this mean it ref to the same pin GP6, right?
```

### Code/REPL

```python
import microcontroller
import board

board_pins = []
for pin in dir(microcontroller.pin):
    if isinstance(getattr(microcontroller.pin, pin), microcontroller.Pin):
        pins = []
        for alias in dir(board):
            if getattr(board, alias) is getattr(microcontroller.pin, pin):
                pins.append("board.{}".format(alias))
        if len(pins) > 0:
            board_pins.append(" ".join(pins))
for pins in sorted(board_pins):
    print(pins)
```

### Behavior

board.BUTTON_L board.BUTTON_R board.GP6
board.GP1 board.LCD_DC board.TX
board.GP10 board.SCK
board.GP11 board.MOSI
board.GP12 board.MISO
board.GP13
board.GP14 board.SDA
board.GP15 board.SCL
board.GP16
board.GP18
board.GP2 board.LCD_CLK board.RX
board.GP21
board.GP22 board.TFT_POWER
board.GP25 board.LED
board.GP7
board.GP8
board.GP9
board.LCD_BACKLIGHT
board.LCD_CS
board.LCD_MOSI
board.LCD_RESET
board.VOLTAGE_MONITOR

tannewt, 2023-05-22:
Looks like an incorrect pin mapping: https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/boards/lilygo_t_display_rp2040/pins.c#L16

I bet BUTTON_R is incorrect.

phamhuyvn, 2023-05-23:
> ###

I use BUTTON_R is normal.
I can't use BUTTON_L same time with BUTON_R , but if I use GP7 instead, 2 button work well

phamhuyvn, 2023-05-23:
{ MP_ROM_QSTR(MP_QSTR_VOLTAGE_MONITOR), MP_ROM_PTR(&pin_GPIO29) },

pin VOLTAGE_MONITOR need to be set as GPIO26 at BAT_ADC , I check pin by this schematic 
https://github.com/Xinyuan-LilyGO/LILYGO-T-display-RP2040/blob/main/schematic/T-Disply%20RP2040_2.pdf

tannewt, 2023-05-23:
The pins.c file will need an update. Want to try making a PR? I don't have this board to test with.

phamhuyvn, 2023-05-24:
> The pins.c file will need an update. Want to try making a PR? I don't have this board to test with.

yes , I can't access GPIO26/Battery voltage and Button_L I can use GP7 instead

phamhuyvn, 2023-05-24:
> The pins.c file will need an update. Want to try making a PR? I don't have this board to test with.

Thanks @tannewt for your support, I PR new edit and now it working well.

closed 2023-05-24 by phamhuyvn