adafruit/circuitpython #8020: Hardware crash involving sdcardio and analogbufio

gpeyton, 2023-05-23:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-alpha.2-132-g3ab9920927

Running on ESP32-S2 (custom board).
```

### Code/REPL

```python
import time
import board
import analogbufio
import array
import supervisor
import asyncio
import gc
import busio
import sdcardio, storage, os

# Mount file system
sd_spi = busio.SPI(clock=board.SCLK, MOSI=board.SD_DIN, MISO=board.SD_DOUT)
sdcard = sdcardio.SDCard(sd_spi, cs=board.CS)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")

# Set up sampling
SAMPLE_RATE = const(22050)
CAPTURE_TIME_SECONDS = const(2)
ADC_PIN = board.MIC_OUT
SAMPLE_SIZE_IN_BYTES = 2

_TICKS_PERIOD = const(1<<29)
_TICKS_MAX = const(_TICKS_PERIOD-1)
_TICKS_HALFPERIOD = const(_TICKS_PERIOD//2)

def ticks_diff(ticks1, ticks2):
    "Compute the signed difference between two ticks values, assuming that they are within 2**28 ticks"
    diff = (ticks1 - ticks2) & _TICKS_MAX
    diff = ((diff + _TICKS_HALFPERIOD) & _TICKS_MAX) - _TICKS_HALFPERIOD
    return diff

def ram_available(sample_rate, capture_time_seconds, sample_size_in_bytes):
    ram_required = sample_rate*capture_time_seconds*sample_size_in_bytes
    gc.collect()
    ram_available = gc.mem_free()
    return (ram_required < ram_available)

async def collect_samples(adc_pin, sample_rate, capture_time_seconds, sample_size_in_bytes):
    try:
        numSamples = sample_rate*capture_time_seconds
        print("Reading",numSamples,"samples from",adc_pin,"pin")
        mybuffer = array.array('H', [0]) * numSamples
        adcbuf = analogbufio.BufferedIn(adc_pin, sample_rate=sample_rate)
        start_time = supervisor.ticks_ms()
        numCapturedSamples = adcbuf.readinto(mybuffer)
        end_time = supervisor.ticks_ms()
        adcbuf.deinit()
        if(numCapturedSamples == numSamples):
            timetaken = ticks_diff(end_time,start_time)
            print("Number of samples read:",numCapturedSamples)
            print("Time required",timetaken,"ms")
            print("Desired sampling rate:",sample_rate,"/s")
            print("Computed sampling rate:",(numCapturedSamples*1000)/timetaken,"/s")
        else:
            print("Unable to capture all samples")
    except (ValueError, RuntimeError) as e:
        print("Exception", e)

def main():
    print("\n=================")
    if ram_available(SAMPLE_RATE,CAPTURE_TIME_SECONDS,SAMPLE_SIZE_IN_BYTES):
        print("Starting code")
        task1 = asyncio.create_task(collect_samples(ADC_PIN,SAMPLE_RATE,CAPTURE_TIME_SECONDS,SAMPLE_SIZE_IN_BYTES))
        await task1
    else:
        print("Not enough RAM available to collect required samples")
    print("=================\n")

asyncio.run(main())
```

### Behavior

Mounting causes an instant hardware crash. Note that the SD card works in all other situations (i.e. when not using `adcbuf.readinto(mybuffer)`), so sdcardio is not the root cause. 

Commenting out the "mount file system" code causes the code to run fine. Likewise, commenting all `analogbufio` code allows SD to mount successfully. So combining them breaks something.

milindmovasha, 2023-05-26:
Hello, This issue was reproduced using ESP-IDF sample application which mounted an SD card and then tried to initialize the DMA engine. 

```
I (287) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (297) sdspi_transaction: cmd=52, R1 response: command not supported
I (337) sdspi_transaction: cmd=5, R1 response: command not supported
I (377) ADC DMA: Filesystem mounted
Name: SU16G
Type: SDHC/SDXC
Speed: 20 MHz
Size: 15193MB
I (377) ADC DMA: TIMES=256

I (377) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (387) spi: SPI3 already claimed by spi master.
E (387) spi: alloc_dma_chan(221): no available dma channel

assert failed: dma_chan_free spi_common.c:321 (spi_dma_chan_enabled & BIT(dma_chan))

Backtrace: 0x40023826:0x3ffc4bd0 0x400287ed:0x3ffc4bf0 0x4002e75d:0x3ffc4c10 0x4008be18:0x3ffc4d30 0x4008c007:0x3ffc4d50 0x4008df70:0x3ffc4d70 0x4008e0e0:0x3ffc4d90 0x40085c84:0x3ffc4dd0 0x40085ee0:0x3ffc4eb0 0x400a4393:0x3ffc4fd0 0x4002acbd:0x3ffc4ff0
0x40023826: panic_abort at /space/projects/esp32-adc-dma/mm/circuitpython/ports/espressif/esp-idf/components/esp_system/panic.c:408 
```

The ESP-IDF ADC DMA APIs uses SPI3 interface for DMA. This interface is already allocated by SD card APIs. While reporting the error an assertion getting hit inside ESP-IDF APIs which would cause hardware crash on CircuitPython.

Ideally ESP-IDF API should return an error here instead of assertion. The assertion is being hit in error path where the API attempts to cleanup before returning an error and asserts in cleanup code where it attempts to free an un-allocated DMA channel.

gpeyton, 2023-05-27:
Not a IDF fix, but a workaround in CP:

```
# Mount file system
dummy_spi = busio.SPI(clock=board.A1, MOSI=board.A2, MISO=board.A3)
sd_spi = busio.SPI(clock=board.D36, MOSI=board.D35, MISO=board.D37)
dummy_spi.deinit()
sdcard = sdcardio.SDCard(sd_spi, cs=board.A0)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
```

closed 2023-05-27 by gpeyton