adafruit/circuitpython #8032: Wifi reconnection does not work on ESP32 and ESP32S3

wuhao19920619, 2023-05-24:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0 on 2023-05-22; VCC-GND YD-ESP32-S3 (N16R8) with ESP32S3

and

Adafruit CircuitPython 8.1.0 on 2023-05-22; ESP32-S3-Box-Lite with ESP32S3

and

Adafruit CircuitPython 8.1.0 on 2023-05-22; Espressif ESP32-EYE with ESP32
```

### Code/REPL

```python
from debugfun import breakpoint, vars
import supervisor
supervisor.status_bar.console = False
supervisor.runtime.autoreload = False

help("modules")

import microcontroller
import board
print(dir(board))
board_pins = []
for pin in dir(microcontroller.pin):
    if isinstance(getattr(microcontroller.pin, pin), microcontroller.Pin):
        pins = []
        for alias in dir(board):
            if getattr(board, alias) is getattr(microcontroller.pin, pin):
                pins.append("board.{}".format(alias))
        if len(pins) > 0:
            board_pins.append(" ".join(pins))
for pins in sorted(board_pins):
    print(pins)

import os
import wifi
print("My MAC addr: %02X:%02X:%02X:%02X:%02X:%02X" % tuple(wifi.radio.mac_address))
print("Connect to %s" % os.getenv("CIRCUITPY_WIFI_SSID"))
print("My IP address is %s" % wifi.radio.ipv4_address)
```

### Behavior

I put my WiFi information in the settings.toml to use web workflows. However, once my router reboot, the Circuitpython can not reconnect to the WiFi anymore.

I have tested ESP32 and ESP32-S3 and the situation is the same.

I tried use REPL to run `print(wifi.radio.ipv4_address)`. There are 3 possible outputs: `0.0.0.0`, `None`, `192.168.4.4`. However my DHCP is `192.168.50.*`，which is not the `192.168.4.4`.

Also, I tried to run `wifi.radio.connect("SSID","password")`, but it only worked when the output is `None`.

For the other 2 outputs, I have to run `wifi.radio.enabled=0`,`wifi.radio.enabled=1`,`wifi.radio.connect("SSID","password")` to make it work again.

anecdata, 2023-05-25:
Can you try it with another router, or with your phone as a hotspot?

Sometimes a router will have a transient `0.0.0.0` IP address for a station as DHCP kicks in. But the `192.168.4.4` is more curious.

wuhao19920619, 2023-05-27:
Hi,

Thank you for your reply.

I figured out where does the 192.168.4.* comes from. It come from a esp8266 running micropython code:
```
from debugfun import breakpoint, vars
print("Activating network")
import network
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect('****', '****')
```

I guess it goes into some kind of AP mode with the same SSID and pwd but not functional.

Also, I guess the Document page need to be modified.
[https://docs.circuitpython.org/en/latest/shared-bindings/wifi/index.html](url)

Under the `connect()` it says "Connects to the given ssid and waits for an ip address. Reconnections are handled automatically once one connection succeeds."

anecdata, 2023-05-28:
The connect comments could be clarified. If the station is disconnected, the code will make several reconnection attempts, but it can't guarantee that the external conditions are ready for a successful connection. Sometimes those retries are enough, sometimes not. And sometimes user code may want to do something else if it can't reconnect, like display an error, or try a different AP.

wuhao19920619, 2023-05-28:
Thank you for helping me sort these things out. If there is not anything confusing I am closing this issue.

closed 2023-05-28 by wuhao19920619