adafruit/circuitpython #8036: Fetch submodules lazily

dhalbert, 2023-05-26:
Instead of having to do a slow `make fetch-submodules` at the top level after cloning `circuitpython`, fetch the submodules lazily. Fetch the submodules for a particular port only when a build is done in that port directory.

... working on this.

bill88t, 2023-05-27:
I feel like `make fetch-submodules` shouldn't be removed.
It's nice to be able to run one command to prepare the local repo so that you can go offline.

But seperating it into auto triggers for every port sounds like a great idea.

joshua-beck-0908, 2023-06-01:
That would be nice. I just wait a long time for every submodules just to change one line of an esp32 board definition. Then the esp-idf build failed anyway. 😅

PR #8070 (2023-06-06): Provide port-specific fetch of submodules
Fixes #8036.

- Rename top-level `make fetch-submodules` and `remove-submodules` to `fetch-all-submodules` and `remove-all-submodules`.
- In each `ports/*`, provide `make fetch-port-submodules`, which fetches only the submodules needed for that port.
- `make fetch-port-submodules` does not require `BOARD` to be set, but most port-specific Makefiles assumed that `BOARD` would always be set. They did not anticipate non-board-specific make targets. Fix that, by fixing how bad or missing `BOARD` is handled, and by guarding various actions in Makefiles to n [... cut]
closed 2023-06-06 by dhalbert