adafruit/circuitpython #8041: Input() block  lead to Internal watchdog timer expired when I2SOut object exist.

wuhao19920619, 2023-05-27:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0 on 2023-05-22; ESP32-S3-Box-Lite with ESP32S3
```

### Code/REPL

```python
import board
import audiobusio

audio = audiobusio.I2SOut(board.I2S_SCLK, board.I2S_LRCK, board.I2S_CODEC_DSDIN)

def breakpoint():
    while True:
        _dbg_input = input('> Input:\n')
        print('> Output:')
        if _dbg_input:
            try:
                exec(_dbg_input)
            except Exception as e: 
                print('> ERROR:', e)
        else:
            break

breakpoint()
```

### Behavior

When there is no tinyuf2 exist (firmware flashed from bin with esptool), if you don't input anything, just let the code pause at `input()`, The ESP32S3 will reset into safe mode within 5 minuates. The reason is Internal watchdog timer expired.

If there is a tinyuf2 exist, it will reset into tinyuf2 bootloader. My project is unattend so unfortunately I have to go there to reset it manually. Because I can not trigger a hard reset from the serial port. (I plug it on the resperberry pi).

If you remove the I2SOut() object, everything will be ok, no matter how long you stay in the input() function.

dhalbert, 2023-06-22:
I tested on an S3 Box Lite using the code above, with CircuitPython 8.1.0. I let the terminal program sit without input as described above for 15 minutes and did not fall into safe mode. When I entered a blank line, the program exited as expected.

Have you been able to reproduce this on any other Espressif boards? I also tried a Metro ESP32-S2, and could not reproduce.

HonestQiao, 2023-07-08:
I tested on an adafruit-esp32-s3-tft-feather using the code above, with CircuitPython 8.1.0.
After running, I waited for over 10 minutes, but nothing happened.
I enter characters and it is able to return.
I enter a blank line and it exits.

HonestQiao, 2023-07-08:
I have a S3 Box Lite.

UF2: tinyuf2-espressif_esp32s3_box-0.15.0

CircuitPython 8.1.0 on an S3 Box Lite , nothing happened after 5 minutes.
Adafruit CircuitPython 8.1.0 on 2023-05-22; ESP32-S3-Box-Lite with ESP32S3

CircuitPython 8.2.0 on an S3 Box Lite , nothing happened after 5 minutes.

skerr92, 2023-11-18:
I'm also having issues with I2SOut causing my esp32-s3 to crash and enter safe mode. I can reproduce it with the code from the MAX98357 learn guide for the sine wave output. There error I get is that the watchdog timer expired.

```py
# SPDX-FileCopyrightText: 2018 Kattni Rembor for Adafruit Industries
#
# SPDX-License-Identifier: MIT

import time
import array
import math
import audiocore
import board
import audiobusio

sample_rate = 8000
tone_volume = .1  # Increase or decrease this to adjust the volume of the tone.
frequency = 440  # Set this to the Hz of the tone you want to generate.
length = sample_rate // frequency  # One freqency period
sine_wave = array.array("H", [0] * length)
for i in range(length):
    sine_wave[i] = int((math.sin(math.pi * 2 * frequency * i / sample_rate) *
                        tone_volume + 1) * (2 ** 15 - 1))

# Using the ESP32-S3-MINI-N8 dev kit flashed from bin
audio = audiobusio.I2SOut(board.IO5, board.IO6, board.IO7)

sine_wave_sample = audiocore.RawSample(sine_wave, sample_rate=sample_rate)

while True:
    audio.play(sine_wave_sample, loop=True)
    time.sleep(1)
    audio.stop()
    time.sleep(1)
```