adafruit/circuitpython #8044: Pi Zero W unable to boot successfully from a fresh 8.1.0 image.

JelleJansen, 2023-05-28:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0 on 2023-05-22; Raspberry Pi Zero W with bcm2835
```

### Code/REPL

```python
print("Hello World!")
```

### Behavior

- The gadget serial interface and CIRCUITPY drive are unavailable. 
- Inspecting the sd card confirms that the 2nd partition was created successfully.
- Content of `boot_out.txt` is as expected and does not provide any clues.

### Description

The console on GPIO 14/15 shows the usual output but seems to hang on  `code.py` :
```Serial console setup
Adafruit CircuitPython 8.1.0 on 2023-05-22; Raspberry Pi Zero W with bcm2835
Board ID:raspberrypi_zero_w
UID:00000000XXXXXXXX
0;🐍Done | 8.1.0
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
0;🐍code.py | 8.1.0
```

Testing intermediate builds since 8.0.5 found these features working up until `8.1.0 beta 2`
- [*] 8.1.0 beta 0
- [*] 8.1.0 beta 1
- [x] 8.1.0 beta 2

### Additional information

This is as much as i could find without digging into binaries.

graeme-winter, 2023-05-28:
I’ve found the same, also from my own build. Four blinks and that’s it.

RetiredWizard, 2023-05-29:
After bisecting it looks like the pystack fix for the espressif chips in commit [7d02bff](https://github.com/adafruit/circuitpython/commit/7d02bff6b8ad473c6789f4b435e3fd64d0f8c78a) is causing this:
```
-    if (stack_get_bottom() != NULL) {
-        mp_stack_set_limit(stack_get_length() - 1024);
+    uint32_t *stack_bottom = stack_get_bottom();
+    if (stack_bottom != NULL) {
+        size_t stack_length = stack_get_length();
+        mp_stack_set_top(stack_bottom + (stack_length / sizeof(uint32_t)));
+        mp_stack_set_limit(stack_length - 1024);
```

graeme-winter, 2023-05-29:
Confirmed @RetiredWizard thank you

With

```
(ENV-circuitpython) graeme@sleeper-service:~/git/circuitpython/ports/broadcom$ git diff
diff --git a/main.c b/main.c
index 122f079129..a32365defe 100644
--- a/main.c
+++ b/main.c
@@ -160,7 +160,6 @@ STATIC void start_mp(supervisor_allocation *heap, supervisor_allocation *pystack
     uint32_t *stack_bottom = stack_get_bottom();
     if (stack_bottom != NULL) {
         size_t stack_length = stack_get_length();
-        mp_stack_set_top(stack_bottom + (stack_length / sizeof(uint32_t)));
         mp_stack_set_limit(stack_length - 1024);
     }
```

board boots 👍

JelleJansen, 2023-05-29:
Cheers! @RetiredWizard 
```
Serial console setup
Adafruit CircuitPython 8.1.0-dirty on 2023-05-29; Raspberry Pi Zero W with bcm2835
Board ID:raspberrypi_zero_w
UID:00000000XXXXXXXX
0;🐍Done | 8.1.0-dirty
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
0;🐍code.py | 8.1.0-dirtyHello World!
0;🐍Done | 8.1.0-dirty
Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```
The hunt for one of those elusive 2w's is still on but i will gladly post a follow-up when luck strikes. :beers:

tannewt, 2023-05-30:
Any idea why it is crashing with that line? Maybe the `uint32_t` needs to be `size_t`?

Without setting the top, CP's stack checking won't be correct.

RetiredWizard, 2023-05-30:
I did some quick test for obvious things like stack_length coming back zero or strange stack_bottom but didn't go very far. I get lost quickly when I start tracing the memory/stack logic but I can give another try :).

Given it doesn't happen on the Zero2W, I'm thinking it might be a difference in how the SDK handles the chip variants???

tannewt, 2023-05-31:
Thanks! I think Zero2 is 64 bit and Zero is 32 bit. That could explain it. They could also have different alignment requirements.

RetiredWizard, 2023-06-07:
@Macrosii I assume you commented out the mp_stack_set_top call to get the board to boot. That fix is likely to result in some instabilities but you could try setting a larger pystack using the **settings.toml** file.

Just add the following line to **settings.toml**:
```
CIRCUITPY_PYSTACK_SIZE=3600
```
If that doesn't help try increasing the size, you probably can go pretty high since the pizero has plenty of RAM.

dhalbert, 2023-06-15:
Making long-term. Will revisit when USB host is working.

vladfr, 2023-08-20:
Just to say that I've tried this on a Raspberry Pi Zero W. The latest 7.x (7.3.3) works fine. But 8.1.0 and 8.2.3 both crash, either don't mount at all, or give out a this error on every run: "NLR jump failed. Likely memory corruption."

RetiredWizard, 2023-08-21:
I added the following print statement to the start_mp function:
```mp_printf(&mp_plat_print, "%d, %d, %d\n", stack_bottom,stack_length,sizeof(uint32_t));```
which printed `4, 524284, 4` on a zero2w and `35623556, 66736, 4` on a zerow.

I'm really not great at c data types and printing the pointer value of stack_bottom feels suspect to me, however..... based on the math being used to calculate the top of the stack, a "stack_bottom" of 35623556 looks like it will cause problems to me.

If it was earlier in the evening I'd track down the stack_get_bottom function and see if I could make heads or tails of it, but it will have to wait until I get a chance next week.

dhalbert, 2023-10-28:
I think this is fixed by #8305. Please re-open if not.

closed 2023-10-28 by dhalbert