adafruit/circuitpython #8046: '/' is mounted READ-WRITE by default on esp32

bill88t, 2023-05-28:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0 on 2023-05-22; M5Stack Timer Camera X with ESP32
```

### Code/REPL

```python
Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 8.1.0 on 2023-05-22; M5Stack Timer Camera X with ESP32
>>> import os
>>> a = open("/test", "w")
>>> a.close()
>>> os.remove("/test")
>>>
```

### Behavior

root is mounted rw

### Description

It could have to do with the web-workflow.

By default, the board should boot to read-only mode.

Neradoc, 2023-05-29:
That's intentional, the only reason to limit the drive to read only is to protect from corruption when the USB drive is mounted.
Builds where USB is not enabled have the drive is read/write mode by default.

bill88t, 2023-05-29:
Thanks for clarifying!

closed 2023-05-29 by bill88t