adafruit/circuitpython #8047: CIRCUITPY drive not appearing with SAM E54 Xplained Pro

tomiahola, 2023-05-29:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0 on 2023-05-22; SAM E54 Xplained Pro with same54p20
```

### Code/REPL

```python
-
```

### Behavior

USB drive appearing instead of CIRCUITPY drive.

### Description

Hello, could someone help me understand why after installing the latest CircuitPython to my SAM E54 Xplained Pro, the "CIRCUITPY" drive does not appear but "USB Drive" appears instead?

dhalbert, 2023-05-29:
Here are some things to check:

- See https://forums.adafruit.com/viewtopic.php?p=838854#p838854, and the whole thread. A user had trouble, and it turned out some board jumpers were set wrong.
- (If I remember right:) A user had trouble with a newer SAMD21 Xplained board. It turns out Microchip changed the flash chip on the board during the chip shortage without much if any notification. Perhaps they did something similar on this board. Currently the board build is only expecting an `N25Q256A` chip. What do the flash chip markings say?
- May be related to #6470.

tomiahola, 2023-05-30:
> Here are some things to check:
> 
> * See https://forums.adafruit.com/viewtopic.php?p=838854#p838854, and the whole thread. A user had trouble, and it turned out some board jumpers were set wrong.
> * (If I remember right:) A user had trouble with a newer SAMD21 Xplained board. It turns out Microchip changed the flash chip on the board during the chip shortage without much if any notification. Perhaps they did something similar on this board. Currently the board build is only expecting an `N25Q256A` chip. What do the flash chip markings say?
> * May be related to [(SAME54)  Volume "Circuitpy" not updated after board reset #6470](https://github.com/adafruit/circuitpython/issues/6470).

Hi, the flash chip markings are: 26F064B

tannewt, 2023-05-30:
Please post a picture of the chip. It'll likely have a manufacturer's logo too.

tomiahola, 2023-05-31:
[image]
Here is a picture of the chip.

tannewt, 2023-05-31:
Thanks! That looks like a MicroChip flash chip instead of micron as the board def says. Want to make a PR adding "SST26VF064B" to the list here: https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/boards/same54_xplained/mpconfigboard.mk#L10 ? That's the full name of the [MicroChip flash](https://ww1.microchip.com/downloads/en/DeviceDoc/20005119G.pdf) (nvm.toml define [here](https://github.com/adafruit/nvm.toml/blob/main/flash/microchip/SST26VF064B.toml).)

tomiahola, 2023-06-01:
Made a PR: https://github.com/adafruit/circuitpython/pull/8054

tannewt, 2023-06-01:
Thank you! I've tweaked it so it'll hopefully build.

dhalbert, 2024-07-22:
Reopening since this sounds like it is still a problem.

eightycc, 2025-04-14:
@tomiahola Please re-test with the build artifact from #10249. Thank you!

PR #8054 (2023-06-01): SAM E54 Xplained Pro: add support for SST26VF064B flash chip
Fixes #8047
closed 2025-04-14 by tannewt