adafruit/circuitpython #8049: ESP32-S3 TouchIn on onboard I2C pin causes hang

dhalbert, 2023-05-30:
https://forums.adafruit.com/viewtopic.php?t=201772

CircuitPython 8.0.5
Adafruit ESP32-S3 Reverse TFT Feather
Trying to use `board.D3` (aka `board.SDA`) as a TouchIn causes a hang.

```py
import touchio

print("")
print("touchio_test.py")

while True:
    touch_pin = touchio.TouchIn(board.D3)
    print(touch_pin.value)
    time.sleep(0.1)
  ```

bill88t, 2023-05-30:
Cannot reproduce exactly on regular esp32, however got a weird crash.
```
Adafruit CircuitPython 8.1.0 on 2023-05-22; M5Stack Timer Camera X with ESP32
>>> import board, digitalio, touchio
>>> a = touchio.TouchIn(board.G13)
>>> a.value
False
>>> a.deinit()
>>> a = touchio.TouchIn(board.G4)
>>> a.value
False
```
This is one of the 3 i2c busses of the board.
On the other one, same deal fine.
On the 3rd one, and specifically pin `board.BM8563_SCL`, which is `pin_GPIO14`, this happened:
```
>>> a = touchio.TouchIn(board.BM8563_SCL)
>>> a.value
False
>>> a.deinit()
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12748
load:0x40080400,len:3024
entry 0x400805dc
Serial console setup

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
Internal watchdog timer expired.
Press reset to exit safe mode.
```

s-light, 2023-05-30:
i just tested it with the REPL:
```python
Adafruit CircuitPython 8.0.5 on 2023-03-31; Adafruit Feather ESP32-S3 Reverse TFT with ESP32S3
>>> import board
>>> import touchio
>>> touch_pin = touchio.TouchIn(board.D3)

```
at this point the ESP32-S3 (in my case a [Adafruit ESP32-S3 Reverse TFT Feather](https://www.adafruit.com/product/5691) )
REPL is unresponsive till i reset it.
the CIRCUITPY disc is still working!

jepler, 2023-06-21:
This may have to do with the powerful pull up on an I2C bus...

s-light, 2023-06-21:
i read somewhere that at initializing the lib tries to measure a base value for *non-touch* state.
maybe this times out in some way because of the pullup? (some sort of out of range / overflow thing?)