adafruit/circuitpython #8051: Neopixel Glitch (Random Colours) Using RP2040

delta1911, 2023-05-31:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.5 on 2023-03-31; Waveshare RP2040-Zero with rp2040
```

### Code/REPL

```python
import board
import neopixel

from adafruit_led_animation.animation.rainbow import Rainbow

# Update to match the pin connected to your NeoPixels
pixel_pin = board.GP6
# Update to match the number of NeoPixels you have connected
pixel_num = 8

pixels = neopixel.NeoPixel(pixel_pin, pixel_num, brightness=0.5, auto_write=False)

rainbow = Rainbow(pixels, speed=0.1, period=2)

while True:
    rainbow.animate()
```

### Behavior

Hi, i have issue with a rp2040, i use rainbow animation suddenly sometime neopixel show a flicker  of random color and continue the animation as normal here is a similar issue with some video examples https://github.com/adafruit/circuitpython/issues/3835

tannewt, 2023-05-31:
How are you powering the neopixels? This can happen if you are using 3.3v logic but powering the pixels at 5v.

delta1911, 2023-05-31:
I am powering it with on board 5v supply...

tannewt, 2023-05-31:
For reliable communication you'll either want to power them at 3v or level shift the data stream up to 5v logic. This is why the [ItsyBitsy has pin 5 be 5v logic](https://www.adafruit.com/product/4888).