adafruit/circuitpython #8053: CPX library doesn't work on CPX (recursion depth hit)

tannewt, 2023-05-31:
https://forums.adafruit.com/viewtopic.php?p=974225#p974225

bill88t, 2023-05-31:
Pystack again?

RetiredWizard, 2023-05-31:
I enabled MICROPY_STACKLESS which resulted in the same "RuntimeError: maximum recursion depth exceeded" message but without the traceback.

I then disabled CIRCUITPY_FULL_BUILD, CIRCUITPY_COUNTIO and CIRCUITPY_BITBANGIO and enabled CIRCUITPY_AUDIOIO and CIRCUITPY_PIXELBUF and the example scripts from the forum post ran without crashing.

I know this isn't a solution but figured it might at least help zero in on the issue.

bill88t, 2023-05-31:
Stackless should never print that message..
Across all the testing I ever did on stackless it never showed it, even in depths >2.000.000.
Something is horribly going wrong for that to show.

RetiredWizard, 2023-05-31:
The CPX has a lot less SRAM (32K I believe) than the chips you were probably testing on, would actually running out of available memory possibly trigger the error?

PR #8071 (2023-06-07): Slim down stack frames
This reduces the stack frame size of mp_builtin___import__ by limiting the support path length of files from 256 to 96. This function can be called recursively for nested imports so it adds up.

Also reduce mp_execute_bytecode (vm.c) from 206 a bc call to 124. This too is recursive and adds up. It is reduced by preventing some inlining. It may decrease performance slightly when importing and unpacking.

Adds two new scripts for debugging. One is used from gdb to print frame sizes in a backtrace. The other prints what pcs use a particular stack offset. This helps find inf [... cut]
closed 2023-06-07 by tannewt