adafruit/circuitpython #8055: Enhancement request: Add support for resuming a wave file from a given point

niccit, 2023-06-01:
While building a simple eBook reader project, I realized if the eBook reader lost power, I would lose the position in the eBook.

I would like to be able to use  tell(), to log how much has been played, and seek() to start at the last logged tell() point to continue the reading journey.

I posted this [question](https://forums.adafruit.com/viewtopic.php?t=201821) in the forums and learned that audiocore.WaveFile doesn't support this.

It would be great if we could add support for beginning a wave file from a designated point.

Please let me know if there is any more information/detail I can provide regarding this request.

Thanks!

tannewt, 2023-06-02:
@jepler any interest in doing this?

samblenny, 2025-05-27:
A useful procedure to learn how a CircuitPython library implementation works is:
1. Find the importable names of the relevant libraries by looking at learn guide code examples or other docs. In this case, for playing wave files, what you probably want is
   ```from audiocore import WaveFile```
   combined with an audio output class like
   ```from audiopwmio import PWMAudioOut as AudioOut```
   or
   ```from audiobusio import I2SOut```
   or
   ```from audioio import AudioOut```
   (see [Play a Wave File](https://learn.adafruit.com/circuitpython-essentials/circuitpython-audio-out#play-a-wave-file-2994862) from the CircuitPython Essentials learn guide, [audioio docs](https://docs.circuitpython.org/en/stable/shared-bindings/audioio/), and [audiobusio docs](https://docs.circuitpython.org/en/latest/shared-bindings/audiobusio/))
2. Check the [shared-bindings](https://github.com/adafruit/circuitpython/tree/main/shared-bindings) and [shared-module](https://github.com/adafruit/circuitpython/tree/main/shared-module) directories in the CircuitPython GitHub repo for the importable names from the previous step. You'll find core library implementations there, but bundle libraries typically have their own repos. For example, you could start by looking at [shared-bindings/audiocore/WaveFile.c](https://github.com/adafruit/circuitpython/blob/main/shared-bindings/audiocore/WaveFile.c), [shared-module/audiocore/WaveFile.c](https://github.com/adafruit/circuitpython/blob/main/shared-module/audiocore/WaveFile.c), and [shared-bindings/audioio/AudioOut.c](https://github.com/adafruit/circuitpython/blob/main/shared-bindings/audioio/AudioOut.c)
3. Use an IDE or code editor to skim through the code in those files to get oriented, then start digging in from a function that relates to what you care about. In this case, the focus is playing wav files, so perhaps you could begin with `audioio_audioout_obj_play()` from the `AudioOut` class implementation in `shared-bindings/audioio/AudioOut.c`.
4. Keep pulling on threads until you find the actual logic for the feature you care about. If your IDE provides a feature to jump to the implementation of a called function, you could use that. Otherwise, multi-file text search features work pretty well. For example, `audioio_audioout_obj_play()` doesn't do much besides handling some housekeeping tasks then calling `common_hal_audioio_audioout_play()`. So, to see the interaction between the implementation of `audioio.AudioOut` and `audiocore.WaveFile`, you need to find where `common_hal_audioio_audioout_play()` is implemented. In this case there are a couple options:
   - [ports/atmel-samd/common-hal/audioio/AudioOut.c](https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/common-hal/audioio/AudioOut.c)
   - [ports/espressif/common-hal/audioio/AudioOut.c](https://github.com/adafruit/circuitpython/blob/main/ports/espressif/common-hal/audioio/AudioOut.c)
   
   Once you read through that stuff, you can see what actually happens for setting up Audio DMA.

So, the challenge would be to find some reasonable way of hooking into the pipeline of buffering audio samples from the long wav file into RAM, then watching how much of that has been shoveled out of a hardware audio interface.

Something to keep in mind is that SAMD21 builds are very, very tight on code space. So, changes that increase the build size on SAMD21 may be hard to get merged.

**Another approach to consider:**

Rather than modifying the CircuitPython core, you could look at ways of using the regular CircuitPython file IO API to prepare a sample buffer with just the samples you need:
1. You could take a look at the [CircuitPython Library Bundle](https://docs.circuitpython.org/projects/bundle/en/stable/) to see if an [existing library](https://docs.circuitpython.org/projects/wave/en/latest/) implements a feature to play only part of a wave file
2. You could use audio tools like Audacity or [`sox`](https://manpages.ubuntu.com/manpages/jammy/man1/sox.1.html) to prepare an audio file with raw samples. Once you have a file of raw samples, you could use CircuitPython's normal iostream `seek()` and `read()` methods to copy a portion of that file into an `audiocore.RawSample` buffer.

samblenny, 2025-05-28:
> what is the difference between a module and binding?

Honestly... no clue, but I haven't found the mystery of it to be an obstacle. Mostly, I just write higher level CircuitPython code. When I go spelunking through the core, I'm usually looking for answers to questions that aren't obvious from the documentation, or else trying to understand quirky behaviors.

The build system and code layout grew organically out of the original stuff from the MicroPython fork 8+ years ago. Everything is the way it is for a reason, or probably several layered reasons. But, it would take a long time to understand all of that. It's easier just to get good at using your editor (or `grep` on the command line) to navigate around the codebase. 

If you want to know what uses files from `shared-module`, you can try this:
```
cd circuitpython
grep -ir shared-module * | grep -v '^shared-module
```
You should get a bunch of `#include ...` lines, stuff from `.mk` makefile includes, and so on. The first grep command is matching against all files that mention "shared-module", the second grep command filters out matching files that are located in the shared-module directory.

todbot, 2025-05-28:
> Quick question that I can't find a definite answer to: what is the difference between a module and binding?

The `shared-bindings` directory contains the code that handles Python-style invocation (e.g. when you call `wav = audiocore.WaveFile()` you're calling [`shared-bindings/audiocore/WaveFile.c`](https://github.com/adafruit/circuitpython/blob/main/shared-bindings/audiocore/WaveFile.c#L55C17-L55C41).  The `shared-binding` is where you do the parsing of Python arguments into C variables and call C functions to do the actual work.

The C code that does the actual work (WAV file loading here) is in `shared-module`, specifically in this case [`shared-module/audiocore/WaveFile.c`](https://github.com/adafruit/circuitpython/blob/main/shared-module/audiocore/WaveFile.c#L32).   And note that this is for port-agnostic modules.  If a module needs chip-specific knowledge (like say `audiopwmio`), then that's in the `common-hal` directory of the corresponding port.