adafruit/circuitpython #8057: CircuitPython IDF Build Error: Git describe was unsuccessful

joshua-beck-0908, 2023-06-02:
### CircuitPython version

```python
N/A
```

### Code/REPL

```python
N/A
Building from the latest Git release.
```

### Behavior

I get the following log when running esp-idf/install.sh :

Detecting the Python interpreter
Checking "python" ...
Python 3.10.6
"python" has been detected
Installing ESP-IDF tools
Current system platform: linux-amd64
Selected targets are: esp32h2, esp32c3, esp32, esp32s3, esp32s2
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32
Skipping xtensa-esp-elf-gdb@11.2_20220823 (already installed)
Skipping riscv32-esp-elf-gdb@11.2_20220823 (already installed)
Skipping xtensa-esp32-elf@esp-2021r2-patch5-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2021r2-patch5-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2021r2-patch5-8.4.0 (already installed)
Skipping riscv32-esp-elf@esp-2021r2-patch5-8.4.0 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20221026 (already installed)
Installing Python environment and packages
fatal: No tags can describe '630c2724fc8c69eeaaa1bb025de52b99c5cb11aa'.
Try --always, or create some tags.
WARNING: Git describe was unsuccessful: b''
ERROR: This script was called from a virtual environment, can not create a virtual environment again

dhalbert, 2023-06-02:
Please try a fresh clone, or do `make remove-submodules` and then `make fetch-submodules` at the top level. Is your clone checked out on `main` or a branch, or something else. (I am looking at the discord discussion as well.)

If it persists, let us know the Linux version and the git version.

joshua-beck-0908, 2023-06-03:
Initially, removing and fetching the modules again did not resolve the problem; however updating my Git version from the distro's version 2.26 to the PPA's version 2.40 and doing the same did.

It seems like this problem is just due to an old version of Git not fetching submodules correctly. 

In future it might be nice to have a better warning of this Git issue, or even a fatal error if an old version is detected, but I think the issue is resolved for now.

closed 2023-06-03 by joshua-beck-0908