adafruit/circuitpython #8059: Wifi ConnectionError Unknown failure 15 on CPY 8.2.0 beta 0

PaulskPt, 2023-06-02:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.0-beta.0 on 2023-05-24; Adafruit Feather ESP32-S2 TFT with ESP32S2
```

### Code/REPL

```python
import wifi
 wifi.radio.connect('...','...')
```

### Behavior

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ConnectionError: Unknown failure 15

### Description

This test done in the REPL window of  mu 1.2.0.

In another testscript with wifi scanning networks results, the transmission I was trying to connect to (above) was reported with:

network ssid: '... ', channel:  4, rssi:   -31

Error might be related to (closed) issue #7127

anecdata, 2023-06-02:
I can't reproduce this on my ESP32-S2 with 8.2.0-beta.0 (or prior). Is connect successful on prior releases on the same board and in the same physical environment (same AP, same distance from AP, etc.)?

If you catch the exception and retry, does it eventually connect? Something like...
```py
import time
import os
import wifi

time.sleep(3)  # wait for serial

print(f" {time.time()} Connecting...")
while not wifi.radio.connected:
    print(f" {time.time()} {wifi.radio.connect(os.getenv("WIFI_SSID"), os.getenv("WIFI_PASSWORD"))} {wifi.radio.connected}", end="\r")
print(f"\n{wifi.radio.ipv4_address}")
```

PaulskPt, 2023-06-02:
Problem resolved: I made a mistake. I mixed the passwords of the two SSID's of the WiFi system (Netgear Orbi Satellites).

soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
 946688072 Connecting...

192.168.1.80

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

@anecdata thank you for your rapid reply.

closed 2023-06-02 by PaulskPt