adafruit/circuitpython #8062: Xiao ESP32C3 usb_hid support

a-fina, 2023-06-03:
Enable support for usb_hid module in order to use HID Keyboard and Mouse

todbot, 2023-06-03:
I believe ESP32C3 has only USB serial capability not full native USB.

bill88t, 2023-06-03:
Yea that.
It can't do hid.
It's serial + jtag only.

dhalbert, 2023-06-04:
@todbot and @bill88t are correct. HID is not possible, nor is MSC, to expose the CIRCUITPY drive. The chip has "native USB" only in the sense that it has on-board USB serial.

closed 2023-06-04 by dhalbert

a-fina, 2023-06-04:
Thanks.

Do you think it should be possible to use the adafruit_hid.consumer_control also without the usb_hid modul?

It seems to be possible looking at this project:
https://learn.adafruit.com/bluetooth-le-hid-volume-knob-with-circuitpython/code-the-ble-volume-knob

dhalbert, 2023-06-04:
The BLE support available for the Espressif chips is quite limited right now, due to a mismatch between how the the underlying software works and what we need fir CircuitPython. The capabilities are explained here: https://learn.adafruit.com/welcome-to-circuitpython/frequently-asked-questions#faq-3129409. So sorry, no, you can't use the BLE support for HID now. If you are interested in HID support over USB, we'd recommend an ESP32-S2 or ESP32-S3 board if you want to stick with a WiFi-capable obard, or almost any other board with native USB, such as an RP2040 board. If you want BLE HID, an nRF52840 is the best choice.