adafruit/circuitpython #8066: Initial pin states after power up (ESP32-S2)

T94T, 2023-06-05:
### CircuitPython version

```python
Official Adafruit CircuitPython v8.1.0
and
Custom compiled Adafruit CircuitPython v8.2.0
```

### Code/REPL

```python
# no custom code (just the factory default code.py file)
```

### Behavior

-

### Description

I have noticed that all pins output 3.3V after a power up. Presumably the internal pull-ups have been activated by default. So I ask here where this is defined in the source code of CircuitPython. Is this intentional or a bug?

_This is my setup and procedure:_
**Hardware**: Lolin S2 Mini (based on ESP32-S2FN4R2)

**Step 1**: Uploaded an "empty" script compiled with arduino-esp32 (v2.0.6) and all pins remained in input state (0V). 
**Step 2**: Burned the TinyUF2 bootloader. The board stays in DFU mode and all pins remain in input state (0 V).
**Step 3**: Uploaded CircuitPython (UF2 binary, v8.1.0 and also custom compiled v8.2.0). After reset and waiting for CircuitPython to fully boot, all pins output 3.3V.

dhalbert, 2023-06-05:
The default pin reset code provided by the ESP-IDF resets the pins with pull-up (instead of floating). According to the documentation, this minimizes power consumption. It also prevents glitches on external pins that might trigger NeoPixels, etc. This is discussed in the comments in #5931.

T94T, 2023-06-06:
@dhalbert, thank you for pointing me to the right direction. This is not good because the board always goes into DFU mode on single reset if `DOUBLE_RESET_RC` has been activated when using the TinyUF2 bootloader (https://github.com/adafruit/tinyuf2/tree/master/ports/espressif#usage). A simple but ugly workaround will be to set the `PIN_DOUBLE_RESET_RC` as an input pin in `code.py` (or `main.py`)

jepler, 2023-06-06:
In CircuitPython espressif port, we have a special case for what Adafruit terms the "double tap pin":
```c
// _reset_pin in common-hal/microcontroller/Pin.c
    #ifdef DOUBLE_TAP_PIN
    // Pull the double tap pin down so that resets come back to CircuitPython.
    pull_down = pull_down || pin_number == DOUBLE_TAP_PIN->number;
    #endif
```
I do not know if this is equivalent to the functionality on this board; there's also `espressif_board_reset_pin_number` which allows a particular board to override how a particular pin is set when not otherwise in use.

T94T, 2023-06-06:
Amazing, thanks so much for the in-depth look at this issue.

T94T, 2023-07-21:
Ok, got it. The `DOUBLE_TAP_PIN` definition needs to be set in the board's `mpconfigboard.h` file. Thank you very much!