adafruit/circuitpython #8067: Can't read analog data properly on ESP32 Wroom 32 u using circuitpython

sanroeeee, 2023-06-05:
### CircuitPython version

```python
Circuitpython for esp32
```

### Code/REPL

```python
'''import time
import board
from analogio import AnalogIn

GasIn = AnalogIn(board.D34)
TempIn = AnalogIn(board.D35)

Vref=1.65
while True:
time.sleep(2)
Vgas=(GasIn.value/4096)3.3
Vtemp=(TempIn.value/4096)3.3
M=32.27
T=29.0Vtemp-18.0
if T<20:
Mc=M((1.3T)+74)
else:
Mc=M((0.4T)+92)
Cx=(1/Mc)(Vgas-Vref)
print(Cx*1000, "ppb")'''
```

### Behavior

Hi! I am using ULPSM sensors from Digikey to measure SO2 concentration with the ESP32 on circuitpython. With the formula on the datasheet, I made the following code i posted above

The problem here is that the output is kinda constant (16-20ppm which is big) no matter what I do with the sensor. Vgas and Vtemp also exceeds the reference voltage of 3.3V which I set on my circuit. I was told that to initialize the analog connection The "D" on D34 should be replaced with "A" however, upon running dir(board), it outputs the one seen on the image attached

as you can see there are only digital pins (D) and no "A" pins so im quite confused as according to internet, all esp32 have analog pins.

I hope some would be able to help me. Thanks in advance.

dhalbert, 2023-06-05:
Which specific board is this?

The "D" vs "A" nomenclature is a holdover from Arduino. It doesn't mean that the "D" pins can't do analog (it only meant that on the original Arduino).

dhalbert, 2023-06-05:
This is being covered in this forum topic: https://forums.adafruit.com/viewtopic.php?t=201964. I'll close this for now; I think it is a support issue.

closed 2023-06-05 by dhalbert

dhalbert, 2023-06-05:
Please post in the forum topic you opened, and please mention exactly which CircuitPython firmware file you are using.