adafruit/circuitpython #8068: 'import wifi' locks up ODroid Go (esp32)

fovea1959, 2023-06-05:
### CircuitPython version

```python
8.1.0
```

### Code/REPL

```python
import wifi
```

### Behavior

Locks up. Full log from REPL:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12748
load:0x40080400,len:3024
entry 0x400805dc
Serial console setup
Adafruit CircuitPython 8.1.0 on 2023-05-22; Hardkernel Odroid Go with ESP32
Board ID:hardkernel_odroid_go
UID:03EA4A4C2202

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 8.1.0 on 2023-05-22; Hardkernel Odroid Go with ESP32
>>> import wifi
```

fovea1959, 2023-06-06:
same problem with 8.2.0-beta.0. After hitting the return on the 'import wifi' in the REPL, REPL locks up, and garbage characters also show up on the ODroid Go screen (see attached pix of screen before and after hitting the enter key). Similar to screen in issue #8069 and #6936.

Before Enter key:
[image]

After Enter key:
[image]

tannewt, 2023-06-07:
Bumping to 8.2 since this looks like a memory bounds issue.

tannewt, 2023-06-12:
Do you have anything in other files (like settings.toml)? I am not able to reproduce this on my odroid go.

tannewt, 2023-06-23:
We can reopen if we hear this is still reproducible.

closed 2023-06-23 by tannewt

fovea1959, 2023-06-25:
settings.toml exists, but is empty. hard to examine anything else, it keeps crashing.

fovea1959, 2023-06-25:
loaded 8.2.0-rc.0. REPL does not echo to console now, can now get through a `import os; os.listdir()` without crashing (but still extraneous screen garbage), and `import wifi` still crashes.

dhalbert, 2023-06-25:
@fovea1959 Are there multiple build versions of the ODroid Go? It's strange @tannewt cannot reproduce this on his sample but it consistently crashes for you. Maybe take a picture of your board that makes the markings on the Espressif module visible, etc.?