adafruit/circuitpython #8069: ODroid Go screen scroll problem?

fovea1959, 2023-06-05:
### CircuitPython version

```python
adafruit-circuitpython-hardkernel_odroid_go-en_US-8.1.0.bin
```

### Code/REPL

```python
import os
os.listdir()
```

### Behavior

reboot freshly imaged ODroid. Go into REPL, control-C
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12748
load:0x40080400,len:3024
entry 0x400805dc
Serial console setup
Adafruit CircuitPython 8.1.0 on 2023-05-22; Hardkernel Odroid Go with ESP32
Board ID:hardkernel_odroid_go
UID:03EA4A4C2202

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 8.1.0 on 2023-05-22; Hardkernel Odroid Go with ESP32
>>> import os
>>> os.listdir()
['.fseventsd', '.metadata_never_index', '.Trashes', 'settings.toml', 'code.py', 'lib', 'boot_out.txt']
NotImplementedError: opcode
>>> 
```

The screen shows a lot of gibberish, but echos the NotImplementedError: opcode and >>> prompt correctly.

tannewt, 2023-06-06:
Please try 8.2.0-beta.0 and post a picture of the screen. Thanks!

fovea1959, 2023-06-06:
8.2.0-beta.0 is worse. Doesn't even finish listing the directory contents before going into the ditch.

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12748
load:0x40080400,len:3024
entry 0x400805dc
Serial console setup
Adafruit CircuitPython 8.2.0-beta.0 on 2023-05-24; Hardkernel Odroid Go with ESP32
Board ID:hardkernel_odroid_go
UID:03EA4A3C0E49

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 8.2.0-beta.0 on 2023-05-24; Hardkernel Odroid Go with ESP32
>>> import os
>>> os.listdir()
['.fseventsd', '.metadata_never_index', '.Trashes', 'settings.toml', 
```
[image]

tannewt, 2023-06-12:
I can't reproduce this on 8.2.0-beta.0.