adafruit/circuitpython #8074: Adafruit MacroPad RP2040 Speaker bug

DavidDerMeteor, 2023-06-07:
### CircuitPython version

```python
Normal 8.0.5 Germany
```

### Code/REPL

```python
from adafruit_macropad import MacroPad
import time
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS
from adafruit_hid.keycode import Keycode
from adafruit_hid.consumer_control import ConsumerControl
from adafruit_hid.consumer_control_code import ConsumerControlCode
from adafruit_hid.mouse import Mouse
#from hid_gamepad import Gamepad
import usb_hid
import board
from rainbowio import colorwheel
import neopixel
from adafruit_mcp230xx.mcp23017 import MCP23017
from adafruit_displayio_sh1107_wrapper import SH1107_Wrapper
import displayio
import busio
import digitalio
import terminalio
import time
import adafruit_imageload
from bongo.bongo import Bongo
from adafruit_display_text import label
from game import Game

disp1 = False
disp2 = False
disp3 = False
disp4 = False
disp5 = False
i2c = busio.I2C(board.SCL, board.SDA)
macropad = MacroPad()
#macropad = MacroPad(rotation=90)
last_position = 0
macropad.pixels.brightness = 0.5
start = 1
#mcp = MCP23017(i2c)
display_sleeper = SH1107_Wrapper(macropad.display)
#bongo = Bongo()
#macropad.display.show(bongo.group)
display = board.DISPLAY

try:
    m = Mouse(usb_hid.devices)
    #gp = Gamepad(usb_hid.devices)
    kbd = Keyboard(usb_hid.devices)
    cc = ConsumerControl(usb_hid.devices)
except:
    None

def rainbow(delay):
    for color_value in range(255):
        macropad.pixels[0] = colorwheel(color_value)
        time.sleep(delay)

macropad.play_tone(200, 0.3)

while True:
    pass

    macropad.encoder_switch_debounced.update()
    if macropad.encoder_switch_debounced.pressed:
        start = start + 1

    if start == 7:
        start = 1

##############################################################################################

    if start == 1 and not disp1:

        macropad.display_image("blinka.bmp")
        disp1 = True

    if start == 1:

        display_sleeper.wake()
        #macropad.display_image("blinka.bmp")
        #macropad.pixels.fill((0, 255, 0))
        macropad.pixels[0] = (0, 70, 80)
        macropad.pixels[1] = (0, 255, 0)
        macropad.pixels[2] = (255, 0, 255)
        macropad.pixels[3] = (255, 0, 0)
        macropad.pixels[4] = (255, 255, 0)
        macropad.pixels[5] = (255, 0, 0)
        macropad.pixels[6] = (255, 0, 0)
        macropad.pixels[7] = (255, 255, 0)
        macropad.pixels[8] = (255, 0, 0)
        macropad.pixels[9] = (255, 0, 255)
        macropad.pixels[10] = (255, 0, 0)
        macropad.pixels[11] = (255, 255, 0)

        current_position = macropad.encoder
        if macropad.encoder > last_position:
            cc.send(ConsumerControlCode.VOLUME_INCREMENT)
            last_position = current_position

        if macropad.encoder < last_position:
            cc.send(ConsumerControlCode.VOLUME_DECREMENT)
            last_position = current_position

        key_event = macropad.keys.events.get()
        if key_event:
            if key_event.pressed:
                if key_event.key_number == 0:
                    macropad.play_file("pew.mp3")
                if key_event.key_number == 1:
                    macropad.play_file("sound.mp3")
```

### Behavior

https://cdn.discordapp.com/attachments/1087450948802191434/1106668331001458761/20230512_214308.mp4

It clicks random some times faster....

https://www.myinstants.com/en/instant/pew_pew/

this file klicks to

### Description

It clicks random some times faster.... and some times slower....

todbot, 2023-06-07:
What exactly is the format of your MP3 files?  CircuitPython can really only do low-bitrate MP3s. For RP2040, I'd stick with mono 64kbps data rate at and 22050 sample rate.

DavidDerMeteor, 2023-06-08:
ok Thanks

closed 2023-06-08 by DavidDerMeteor