adafruit/circuitpython #8082: Solve possible long-lived allocator issue vs synthio

jepler, 2023-06-13:
### CircuitPython version

```python
8.2.0-beta.1
```

### Code/REPL

```python
# see forum thread
```

### Behavior

when some synthio objects (particularly the Synthesizer itself?) are imported, synthio breaks.

these objects refer to each other in complicated ways, e.g., AudioOut -> Synthesizer -> Note -> LFO

If one of these objects is moved (made long-lived) it would break the relationships.

At the same time, Notes in particular are not conceptually "long lived", so the usual thing we've done (e.g., with UART, which frequently is pointed to by an interrupt handler object) of initially allocating as long-lived is not a great solution.

This may be a cause of https://forums.adafruit.com/viewtopic.php?t=202163 -- I saw and ignored something similar during testing.

A workaround that still lets the "create audio player", "create synthesizer" object etc code to be re-used is to place a *FUNCTION* for creating the object in the import, instead of creating the object itself there.

todbot, 2023-06-14:
Looks like the issue noted in the forum post has been resolved so I think this issue could be closed?

jepler, 2023-06-14:
I'd like to leave it for now. I was wrong about it being the cause of the user's problem, but there's a real problem here which can arise when you do something like
```py
# go_audio.py
speaker = I2SAudioOut(...)
synth = Synthio(...)
speaker.play(synth)
``

```py
# code.py
from go_audio import synth
synth.press(...)
```
which I believe is because of long-lived objects; the "synth" that is played on "speaker" is not the same "synth" that is imported from code.py, and so things don't work.

jepler, 2023-06-21:
This is a facet of https://github.com/adafruit/circuitpython/issues/2687

dhalbert, 2023-10-27:
Fixed by https://github.com/adafruit/circuitpython/pull/8281, which removed long-lived allocations, and so removed objects moving around after creation.

closed 2023-10-27 by dhalbert