adafruit/circuitpython #8086: Invalid wifi password

caizyy, 2023-06-15:
### CircuitPython version

```python
8.1.0
```

### Code/REPL

```python
import time
import ssl
import socketpool
import wifi
import microcontroller

# wifi.radio.connected

network = {"SPARK-YK3MAZ": "2XSGVxdwd9JDF"}
networkName = list(network)[0]#first network
networkPassword = network.get(list(network)[0])
print("connecting to "+ networkName)

connectionAttemptCounter = 0
while True: # connect to wifi
    try:
        wifi.radio.connect(networkName, networkPassword, timeout=10.0)
        if wifi.radio.connected:
            print("Connected")
            break
        print("ummm idk how dis happen qwq ERROR CODE: 0001")# shouldnt happen
        connectionAttemptCounter = connectionAttemptCounter + 1
    except ConnectionError as e:
        print(e)
    connectionAttemptCounter = connectionAttemptCounter + 1
    if connectionAttemptCounter < 14: #if its 15 reset device
        microcontroller.reset()
```

### Behavior

It crashes with Unknown failure 15

### Description

happens when i type the wrong wifi password, instead it should throw error

dhalbert, 2023-06-15:
Which board are you using?

caizyy, 2023-06-15:
esp32 devkit

anecdata, 2023-06-15:
Does it catch the exception repeatedly, then reset? ...or does it eventually connect? Failure 15 is `WIFI_RADIO_ERROR_4WAY_HANDSHAKE_TIMEOUT`. I recall it has occurred before if signal strength is poor, or also if there are more than one AP device with the same SSID (but different password). There are probably other triggers for that exception. Can you give more detail on the network environment (distances, number of APs, channels used, signal strengths if known from other connections, etc.)?

caizyy, 2023-06-15:
i put a invalid wifi password and its not throwing a error

caizyy, 2023-06-15:
it just crashes

anecdata, 2023-06-15:
Isn't `Unknown failure 15` a `ConnectionError` that's caught by the `try` / `except`? Can you post whatever (if anything) gets printed to the serial console when this happens?

srolf, 2023-06-16:
Maybe it's not related to this issue, but let me share an observation:

I'm using an Wemos D1 R32 with the CircuitPython firmware 8.1.0 for doit_esp32_devkit_v1.
This was working fine for two days with code with utilizes a SS1306 on I2C to show its IP address.
In addition the on-board LED is used for blinking each 0.5sec.
The complete code section for I2C, display and wifi IP address is enclosed by try/catch. So even if it fails because of an I2C issue or no IP address it would start blinking.
Already a second time I see now a strange behaviour:

- After applying power to the board (no USB serial connection, just plain 5V)
- Unexpectedly neither the display shows some text nor the on-board LED starts blinking
- On first occurrence, I tried around by a complete erase and flashing the fw again plus setting up the toml config -> no change
- But by removing the D1 R32 from my list of allowed WiFi device on the router and adding it again, it started working
- On second occurrence, I tried directly just remove/add to my router -> surprisingly no change

By checking the output of D1 R32 via USB serial, in both cases it got stuck directly after showing the UID.

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12748
load:0x40080400,len:3024
entry 0x400805dc
Serial console setup
Adafruit CircuitPython 8.1.0 on 2023-05-22; ESP32 Devkit V1 with ESP32
Board ID:doit_esp32_devkit_v1
UID:<removed here to hide the real MAC>

srolf, 2023-06-18:
Similar behaviour ~24h later.

Meanwhile I modified the code in a way that no longer a sleep in my implementation's while-loop is used, but a check with time.monotonic(). The idea was that maybe the WiFi handling is blocked and D1 R32 and the router get in trouble (because of an expired lease). But this idea seems to be incorrect.

Next step was to restart the router by its web interface.
**Remarkable**
In the moment the restart sequence starts, the D1 RS32 get unblocked from "UID: ..." and continuous with "Auto-reload is on."
Looks for me that something in the **CPy firmware waits forever without timeout to connecting to the given WiFi AP**.

In the meantime the router is fully restarted. If I now restart the D1 R32 by hard reset, it connects to the WiFi AP, gets an IP and shows it properly.

I don't know if the loss of connectivity is because of my router or the CPy firmware, but the blocking behaviour looks for myself caused by a missing timeout in the CPy firmware.