adafruit/circuitpython #8088: Adafruit Grand Central M4 Express: support for canio

srolf, 2023-06-16:
I really like the Adafruit Grand Central M4 Express by using CircuitPython on it. Its ATSAMD51 has got an integrated CAN controller and related pins are exposed. So is there a chance to support **canio** (comparable to an ESP32)?

anecdata, 2023-06-16:
SAM E series has CAN, but I don't think SAM D does

p.2 https://ww1.microchip.com/downloads/aemDocuments/documents/MCU32/ProductDocuments/DataSheets/SAM-D5x-E5x-Family-Data-Sheet-DS60001507.pdf

srolf, 2023-06-17:
The [pinout ](https://learn.adafruit.com/assets/115366)of the Grand Central shows CAN0 and CAN1.

anecdata, 2023-06-17:
I think that's an error from some combined SAM D/E document. The product page for that processor says
> Number of CAN Modules        0

https://www.microchip.com/en-us/product/ATSAMD51P20A

The two Adafruit MCP2515 boards (FeatherWing and Pico PiCowbell) are designed for 3.3v microcontrollers, but the CAN `L` and `H` are:
> 5V as required by the CAN Bus transceiver

I'd be a little surprised if the SPI speed is a bottleneck in an application. But true, the CAN-FD transceivers can operate at 5Mbps whereas the MCP2515 operates at <=1Mbps.

srolf, 2023-06-17:
Thank you very much for the clarification.

Meanwhile I understood the trouble: Adafruit's pinout was misleading me (this [one](https://www.adafruit.com/product/4759) might have been the better option for me). You are correct, the data sheet shows the difference between SMD51 and SME51. The hint about a feather shield is good, maybe I will be able to get hands on [Adafruit CAN Bus FeatherWing - MCP2515](https://www.adafruit.com/product/5709).

closed 2023-06-17 by srolf